pico-coder/sigrok-pico #4: New user: failing to get sample data

DeeJay, 2022-03-01:
Greetings.

This is forum member _'B.Goode'_ from the Raspberry Pi forums.

Raspberry Pi host platform is an RPi4B/2 running RasPiOS Bullseye [64].

I have built sigrok-cli against the version of libsgrok containing your patch.

I have flashed your uf2 to a Pico.

The --list-serial and --scan operations seem to be successful.

But an attempt to get --samples fails.

Sample output -

```
pi@rpitest64:~ $ sigrok-cli --list-serial
Available serial/HID/BT/BLE ports:
  /dev/ttyACM0	Pico - E660583883443A2F
  /dev/ttyAMA0	ttyAMA0

```
```
pi@rpitest64:~ $ sigrok-cli -l 2 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --scan
sr: srgn: Drain reads
sr: srgn: Drain reads done
sr: srgn: sr_err level logging enabled
sr: srgn: sr_warn level logging enabled
The following devices were found:
raspberrypi-pico - Raspberry Pi PICO 00 [S/N: N/A] with 24 channels: A0 D2 A1 D3 A2 D4 D5 D6 D7 D8 D9 D10 D11 D12 D13 D14 D15 D16 D17 D18 D19 D20 D21 D22

```

```
pi@rpitest64:~ $ sigrok-cli -l 5 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --config samplerate=100 --channels D2 --samples 10
sr: [00:00.000002] log: libsigrok loglevel set to 5.
sr: [00:00.000176] backend: libsigrok 0.6.0-git-a72624da/4:0:0.
sr: [00:00.001010] backend: Libs: glib 2.66.8 (rt: 2.66.8/6608:8), zlib 1.2.11, libzip 1.7.3, minilzo 2.10, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.24.11584 API 0x01000108.
sr: [00:00.001192] backend: Host: aarch64-unknown-linux-gnu, little-endian.
sr: [00:00.001399] backend: SCPI backends: TCP, RPC, serial, USBTMC.
sr: [00:00.002091] backend: Firmware search paths:
sr: [00:00.002231] backend:  - /home/pi/.local/share/sigrok-firmware
sr: [00:00.002290] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.002346] backend:  - /usr/share/fkms/sigrok-firmware
sr: [00:00.002397] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.002451] backend:  - /usr/share/raspi-ui-overrides/sigrok-firmware
sr: [00:00.002502] backend:  - /usr/share/sigrok-firmware
sr: [00:00.002553] backend:  - /usr/share/gdm/sigrok-firmware
sr: [00:00.002602] backend:  - /var/lib/menu-xdg/sigrok-firmware
sr: [00:00.002744] backend: Sanity-checking all drivers.
sr: [00:00.002833] backend: Sanity-checking all input modules.
sr: [00:00.002889] backend: Sanity-checking all output modules.
sr: [00:00.002966] backend: Sanity-checking all transform modules.
sr: [00:00.022222] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi (nil) cg NULL -> [uint32 20000, 20001]
sr: [00:00.022367] srgn: Opening /dev/ttyACM0.
sr: [00:00.022416] serial: Opening serial port '/dev/ttyACM0' (flags 1).
sr: [00:00.028391] serial: Parsing parameters from "115200/flow=0".
sr: [00:00.028998] serial: Got params: rate 115200, frame 8/0/1, flow 0, rts -1, dtr -1.
sr: [00:00.029061] serial: Setting serial parameters on port /dev/ttyACM0.
sr: [00:00.029178] serial: DBG: serial_set_params() rate 115200, 8n1
sr: [00:00.029397] serial: Flushing serial port /dev/ttyACM0.
sr: [00:00.029838] srgn: Reseting device with *s at /dev/ttyACM0.
sr: [00:00.030227] serial: Wrote 1/1 bytes.
sr: [00:00.040653] srgn: Drain reads
sr: [00:00.141194] srgn: Drain reads done
sr: [00:00.141306] serial: Wrote 2/2 bytes.
sr: [00:00.144698] serial: Read 17/17 bytes.
sr: [00:00.144733] srgn: num channels a 3 d 21 bps 6 dsb 3
sr: [00:00.144788] srgn: Setting serial buffer size: 256000.
sr: [00:00.144815] srgn: At get_dev_cfg
sr: [00:00.144861] serial: Wrote 3/3 bytes.
sr: [00:00.145953] serial: Read 7/20 bytes.
sr: [00:00.156139] srgn: A0 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: [00:00.156256] serial: Wrote 3/3 bytes.
sr: [00:00.157371] serial: Read 7/20 bytes.
sr: [00:00.167495] srgn: A1 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: [00:00.167580] serial: Wrote 3/3 bytes.
sr: [00:00.168742] serial: Read 7/20 bytes.
sr: [00:00.178875] srgn: A2 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: [00:00.178936] srgn: sr_err level logging enabled
sr: [00:00.178954] srgn: sr_warn level logging enabled
sr: [00:00.178972] srgn: sr_info level logging enabled
sr: [00:00.178989] srgn: sr_dbg level logging enabled
sr: [00:00.179093] srgn: sr_spew level logging enabled
sr: [00:00.179132] hwdriver: Scan found 1 devices (raspberrypi-pico).
sr: [00:00.179296] device: raspberrypi-pico: Opening device instance.
sr: [00:00.179327] serial: Opening serial port '/dev/ttyACM0' (flags 1).
sr: [00:00.180092] serial-libsp: Error opening port (16): Device or resource busy.
Failed to open device.
pi@rpitest64:~ $ 

```

Group and device permissions look OK, and the relevant port seems to get opened correctly earlier in the process?

```
pi@rpitest64:~ $ ls -al /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 Mar  1 16:59 /dev/ttyACM0
pi@rpitest64:~ $ groups
pi adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi
pi@rpitest64:~ $ 

```

Anything I can try, or additional information needed. This is a 64-bit version of RasPiOS Bullseye - do you need a comparison run on the 32-bit version of that OS?

pico-coder, 2022-03-01:
I run that exact command line and it works find.
I think I've seen that issue when I had left a different instance of the application open such that one app had tied up the serial port and then a new one started and couldn't get control of it.  But I've only seen that running pulseview where I forgot to close a window and then started a 2nd session.  Since you are just running sigrok-cli which terminates the connection that doesn't seem to explain it.

It may have something to do with how the serial port is opened/closed multiple times within the cli and maybe there is something OS specific going on.

You might try unplugging/replugging the PICO and/or maybe reseting your host, but I can't say I have  a high confidence that will help.

Let me have a look around the sigrok code and see what I can find... but give a replug a shot to see what happens...

pico-coder, 2022-03-01:
ok, looks like the other devices include a serial_close as part of the original device scan.

can you please try a fix for me?

In libsigrok/src/hardware/raspberrypi-pico/api.c, at the end of the *scan function go from this:
        sr_spew("sr_spew level logging enabled");
        return std_scan_complete(di, g_slist_append(NULL, sdi));

To this:
        sr_spew("sr_spew level logging enabled");
        sr_dbg("Scan closing serial port");
        serial_close(serial);
        return std_scan_complete(di, g_slist_append(NULL, sdi));

If I do that the sigrokcli still works, so I'm guessing that in my NOOBs OS it doesn't mind if the same application calls open twice, but your version might care.

DeeJay, 2022-03-02:
Fixed!!

Explicitly CLOSEing the port after the preliminary probe seems to have done the trick.

'Good' traces from sigrok-cli, with verbose logging disabled -
```
pi@rpitest64:~ $ sigrok-cli -l 0 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --config samplerate=10000 --channels D2 --samples 8
libsigrok 0.6.0-git-a72624da
Acquisition with 1/24 channels at 10 kHz
D2:00000000
pi@rpitest64:~ $ sigrok-cli -l 0 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --config samplerate=10000 --channels A1 --samples 8
libsigrok 0.6.0-git-a72624da
Acquisition with 0/24 channels at 10 kHz
A1: 514.0 mV
A1: 591.1 mV
A1: 616.8 mV
A1: 642.5 mV
A1: 668.2 mV
A1: 668.2 mV
A1: 693.9 mV
A1: 693.9 mV
pi@rpitest64:~ $

```
(Tested with 'bare' Pico which had all pins left floating. Real data acquisition stillll be be checked.)