pico-coder/sigrok-pico #5: What driver with PulseView?

tony1tf, 2022-03-03:
I uploaded the uf2 file into the Pico, and can see the Pico serial interface from PulseView, but I need the name of a driver in order to connect. I havn't found anything which works.

pico-coder, 2022-03-03:
Tony, my apologies.  I've been too busy trying to get a pull request to mainline sigrok that I didn't put a warning in Readme.md that mainline sigrok builds do not support the PICO yet.

I just updated Readme.md to reflect that.

If you are on linux, it is reasonably straightforward to build libsigrok and sigrokcli (see SigrokBuildNotes.md).  
Note that while I have built pulseview on a raspberyy pi model 3B 1GB RAM, I can not recommend it as it will run out of memory and swap to the SD card, possibly taking 8 hours or more, so if you try it on linux do it on something with more RAM.
I have not been successful in building windows, and don't intend to try again.

I will probably keep this open, just so other people see it (in case the bold letters in Readme.md aren't obvious :) ).

tony1tf, 2022-03-03:
Hi Pico-Coder

Thanks for your fast response. I have PulseView on both a Mac Book and a PC
and could also try a RPi400. I didn't realise the complexities of this - I
assumed that your code for the Pico created a standard Sigrok protocol
which any PulseView could use. Currently I have been using a Cypress USB
FX2 clone with the fx2law driver, but I know that has to load code into the
FX2 at startup. I guess I should read the wiki to understand how all this
is supposed to work in order to help.
And your Bold text on the readme file is now very obvious (apart from the
typo!).
Tony

tony1tf, 2022-03-03:
Hi again

My colleague Jeremy who writes comprehensively here:   https://iosoft.blog  has written a logic analyser for the Raspberry Pi which used the already existing Hameg HMO Sigrok driver for communication via TC/PIP. This way he didn't have to get involved with the intricacies of writing a new Sigrok driver. Although I see you have already done a lot of work there. He has been working recently on using the Pico, but with an ESP32 wifi interface.  I suggest he would be a good person for you to converse with and swap notes. I have used the logic analyser code he wrote in 'C' for the R Pi, and it works, but the whole Sigrok/PulseView setup is a bit delicate, and he wasn't happy with it. If you need to copy the code he has written for the Pi, I'm sure he would be happy about that, so I will email him if that is the way to go, but I'll leave it up to you for now.

Tony

pico-coder, 2022-03-03:
Tony,
Thanks a ton for that link.  I've got a different project in mind where I was trying to figure out how to wirelessly transfer some info from a PICO to display on my smart phone and it looks like he's solved the problem in at least one way. I'm good with a soldering iron, but helpless with Java.

Agreed with the delicacy of trying to leverage drivers.  Many of these logic analyzer devices have lots of intricacies and/or limited protocol uses that allow a driver to work just fine on the specific hardware but be very flaky when you try to reverse engineer it.  I actually started to design this project to look like a SCPI device, mostly because it seemed like a cool retro thing to do.  But you quickly learn the protocol and the sw stacks wrapped around have a lot of unintended limitations.

For the sake of this project, I'll say Jeremy's stuff is super interesting, but not directly applicable, mostly because it requires a 2nd intermediate device.

His DMA use looks good, but I don't think it implements the ping-pong setup I implemented here.  Specifically in that setup it allows you to store samples to half of your buffer via DMA, and then as you transmit that half the DMA engine is writing to the other half.  Assuming your link can keep up, that can provide "infinite" trace depth and ensures that you don't have any blips or pauses in the sampling because the DMA and ADC/PIO fifos guarantee a continuous transfer.

FWIW, I suppose you could also leverage the PICO code from this repo to get the DMA/ADC/PIO configurations but send the sample data out wifi and parse it in a browser setup like Jeremy's.  An interesting idea, but one I don't have the free time for :).

Again, nothing against his setup, it's got a lot of good merits. For sure he'd be able to add new features to the wave viewer a lot faster :).

It also wouldn't surprise me if the data bandwidth through that wifi module might be noticeably higher than a usb serial CDC implementation, so he might have a fundamental advantage there .

Shawn

tony1tf, 2022-03-03:
Hi again

I'm not sure how gitHub and pull requests work. Does this mean that at the moment, I won't be able to get an executable for my Mac or PC until someone higher up the chain approves it?

Tony

pico-coder, 2022-03-03:
Correct.  Sigrok supports regular nightly builds at https://sigrok.org/wiki/Downloads, but only for code in their repo.
Their official repo is  git://sigrok.org/libsigrok.
They use github as a read only copy of their main repo, but support github pull requests to it, so you submit a pull request and the maintainers review it and approve it and magically it crosses over.
How that all works I have no idea.

Oh, and while you are reading, what is the typo in Reade.md you mentioned?

tony1tf, 2022-03-04:
communiate!

tony1tf, 2022-03-04:
Hi @pico-coder  You seem to have got your code merged. When does the nightly build take place?

pico-coder, 2022-03-04:
Tony,
I'm not sure why you think that it got merged.

https://github.com/sigrokproject/libsigrok/pull/181 is the pull request and it's not marked as approved or merged.

And if you go to the source directory the raspberry pi pico directory isn't there.
https://github.com/sigrokproject/libsigrok/tree/master/src/hardware

Shawn

tony1tf, 2022-03-04:
Hi Shawn

I saw 'merged' somewhere and I guess it confused me! 
Who do we email to ask them to do it?

Tony

pico-coder, 2022-03-04:
The three approved ways to ask are:
1) Post to the email reflector - Done.
2) Post to chat - Did that too.
3) Post the pull request to github - and did that too.

These are volunteers and handle a large amount of change requests (note the pending pull request count is around 52) , so just need to be patient.

tony1tf, 2022-03-04:
Hi @pico-coder
There only seems to be https://github.com/uwehermann named for the maintenance of the repository - perhaps I am not understanding that?
Tony

pico-coder, 2022-03-04:
I'm no expert , maybe that is the only maintainer, but there might be others that do prequal checks etc.  But, again, I really don't know....

tony1tf, 2022-03-10:
I'm still trying to create PulseView on my Mac with your revised code, but the wiki says I need to 
git clone git://sigrok.org/sigrok-util
So, if I change to 
git clone https://github.com/pico-coder/sigrok-util
it wants my user name and password, then says I need a personal access token. I generated one, but don't know where to put it.
Help - really not getting anywhere with this!

pico-coder, 2022-03-10:
Your personal access token is used in place of your password. I never needed the token until I was trying to do pull requests to github, but that was all on linux.

tony1tf, 2022-03-10:
Thanks @pico-coder. I used that instead of my password, and it got a little farther, but then tole me that the repository doesn't exist, so I still doing something wrong:
Username for 'https://github.com': tony1tf
Password for 'https://tony1tf@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/pico-coder/sigrok-util/' not found

Next hint, please.?

pico-coder, 2022-03-10:
Sorry, my bad that I didn't see this earlier.
I originally wrongly thought that part of the entire sigrok check in flows would require a checkin to sigrok-util and thus I created my own copy of it.  However that isn't the case, you need to checkout the mainline sigrok-util and run the new driver script.  Since there was nothing to checkin I deleted the repo.
I'll update the build notes.

tony1tf, 2022-03-13:
Hi @pico-coder 

Thanks for managing to cross compile a windows version.
I have installed your version of PulseView on a W10 laptop that I usually use for PulseView with an FX2USB device.
I'm using a new Pico and have uploaded your uf2 file to it.
On 'connect to device' PulseView correctly shows the Raspberry Pi Pico driver, and I can see the serial interface (com5). 
I have tried all the different baud rates and the blank one, but it doesn't detect the Pico. 
The log file generated at log level 5 shows:
sr: srgn: Opening COM5.
sr: serial: Opening serial port 'COM5' (flags 1).
serial-libsp: Error opening port (5): Access is denied..
sr: srgn: 1st serial open fail

So it looks like we have a permissions problem. Any thoughts?

Tony

pico-coder, 2022-03-13:
That often means that the serial port has been opened by something else.
Possibly a Virtual machine, or terraterm/putty or perhaps a 2nd instance of
pulseview that is open.

The baud rate doesn't matter because because there is no real serial port.

You might also be missing some kind of windows serial/usb driver. You might
check out the windows help section of the main sigrok page (I put a link to
it in the readme.md under the windows directory of my repo).

Finally, try closing all the apps, unplugging the pico,wait a bit, replug
it, and then open pulseview (it may help things to establish the serial/CDC
before pulseview is opened).

Good luck.

tony1tf, 2022-03-13:
The device manager says the serial interface com5 is operating correctly.
I also tried running pulseview in administrator mode. I'm not getting the permissions error now, but it still isn't working:

sr: srgn: Drain reads
sr: srgn: Drain reads done
srgn: ERROR:Serial_w_resp failed (0).
sr: srgn: 1st identify failed
sr: srgn: Send second *
srgn: ERROR:Serial_w_resp failed (0).
sr: srgn: Second attempt failed

pico-coder, 2022-03-13:
The drain reads indicates that there were some pending characters sent from
the device, likely from a previous command from the host/pulseview.

Replugging the device and restarting the app might help...  I actually saw
this myself and started up a putty session and it went away.  I didn't
think much of it, but maybe there is a different handshake I need to try in
the code...  I'll have a shot when I get home.

tony1tf, 2022-03-13:
Having not been able to get PulseView to recognise the Pico, I used Putty on COM5 at 57600, and got this back from the Pico:
SRPICO,A031D21,00SRPICO,A031D21,00

Tony

pico-coder, 2022-03-13:
So I've reproduced the issue and it appears that if you run Putty that it configures the serial port in such a way that pulseview then starts working (but must always have only one of those open at a time or they will conflict with a serial port).

I did some experimenting with the windows "mode" command in cmd window and this seems to do the trick to set DTR=on.
So the sequence:
1.Close all serial apps
2.Unplug PICO
3.replug PICO
4.open a cmd window
5.run this cmd, using the number of your serial port
mode COM7 BAUD=115200 data=8 Parity=n DTR=on
6.run pulseview.

Long term I'll need to figure out the right settings in the libserial call from pulseview to get DTR set correctly, but let me know if the mode command workaround helps.

tony1tf, 2022-03-13:
Hi

I got it to work once after looking at it with Putty - PulseView showed the digital and analog channels.
I tried your terminal screen setup and that was more consistent, although lock-up seems very easy to do..
It would be nice to have a drop down menu for the acquisition rate since the up down buttons only change by 1Hz. It did seem I was able to type more noughts into the rate, but I havn't got a known input yet, so I wasn't sure what it was doing.  Progress at last!

pico-coder, 2022-03-14:
I just released a new pulseview to Dropbox which should set the DTR rate correctly.  I was able to use pulseview without putty/mode.
The reason for the acquisition rate being as it is is due to how pulseview works.  You can either provide a list of frequencies the user can pick (which would be huge because there are practical reasons to pick many values between 5khz and 120Mhz), or you get what is there now.  Note that you can just type in the frequency you want, rather than clicking 120 million times to reach 120Mhz.  :).
Please file a new issue an any "lockups", especially include information on enabled channels, sample rate and number of samples.

pico-coder, 2022-03-14:
oh, also, regardling the "lockups" make sure you aren't setting the lower decimal value to non-zero.  If you do that enables hardware based triggering - see "HW triggered via PIO" in the AnalyzerDetials.md.
(I know, it's really hacking to make bits of the sample rate control triggering, but pulseview doesn't (AFIK) have a clean way to send user "metadata" from the pulseview gui to the device).

pico-coder, 2022-04-17:
This issue has covered lots of topics, but with the release of rev2 I think everything has been addressed.

closed 2022-04-17 by pico-coder