pico-coder/sigrok-pico #12: Sample rate pulldown redefinition

pico-coder, 2022-03-28:
I originally setup the driver to use the mode where  you could type in the frequency because it would give the most flexibility.
But, nobody likes it (including myself), because:
1) It seems to take lots of key presses to get it to hold.
2) It's hard to keep track of all the zeros when typing in high rates.

However, I don't think the 1x,2x,5x,10x (which you get if you specific min,max,step=2) is fine enough granularity and it misses some key integer dividers.

However, pulseview does allow a complete list. I think this should cover it, but let me know if I'm missing something.
In the >10Mhz range, cover all the integer dividers of 120Mhz, so
10,15,20,30,40,60,120Mhz
In the 1Mhz range, same idea, but even more granularity because we are in the range where rle might be able to enable streaming for various usages. (I might add some more if pulseview supports non integer values in an easy to ready fashion).
1,2,3,4,5,6,8,10 Mhz 
For the 100k range do a similar range including the 1,2 and 3 ADC channel maximums,
100,167,200,250,300,400,500,600,800
For the 10k
10,20,30,40,50,60,80
And under 10k, with the 5k as the minimum
5,6,8,10k

That's 35 frequencies which I think shouldn't be to annoying to scroll through, and as a plus pulseview prints them in SI units so it's easy to read.

Did I miss any?

tony1tf, 2022-03-30:
Sounds good to me - so presumably that's a new Windows build you will upload?

pico-coder, 2022-04-17:
rev2 upload has this.  I added a few other frequencies that were close to the limits where digital RLE can be useful.

closed 2022-04-17 by pico-coder