pico-coder/sigrok-pico #14: 10:1 attenuator on analogue channel

tony1tf, 2022-04-08:
Hi

A question -I want to use a 10:1 attenuator on the analogue channels. I would like the voltage displayed correctly in PulseView.  Am I right in believing that all that is needed is a change in pico_sdk_sigrok to line 163 in srdevice.h from 25700 to 257812 (= 33V/128 uV)

Tony

pico-coder, 2022-04-08:
Yes, that's it.
However, Pulseview has an "add math signal" (look for the f(x) in the menu bar, that allows you to manipulate channel values.  That might be a bit faster if you need to swap attenuators :).

tony1tf, 2022-04-11:
Hi Shawn
Thanks for the suggestion.
I tried the f(x) signal - it's OK, but you can't disable the original channel, else it can't pick it up to be multiplied by 10, so it wastes screen space, which is pretty special when you could have 22 digital and 3 analogues.
Tony

pico-coder, 2022-04-17:
Closing this out because the question is answered, happy tracing.

closed 2022-04-17 by pico-coder