pico-coder/sigrok-pico #15: Tests on version 2

tony1tf, 2022-04-18:
Hi pico-coder

Been doing some tests on your version 2. I gather that there is now no hardware trigger, so the result depends on the speed of the host computer. I'm using an old Fujitsu laptop with P6200 Pentium 2 X 2.13GHz + 4GB RAM
With all 21 dig channels enabled, although only D2 fed with some pulses, and trigger on D2.
1M samples at 2MHz = 500ms acq time = OK
200K at 2.5MHz = 80ms = OK
1M at 3MHz = 333ms = OK
100K at 4MHz = 25ms = abort
With no trigger and 8 ch enabled:
200K at 4MHz = 50ms = OK
500K at 4MHz = 125ms = abort
200K at 120MHz = 2ms = OK
No trigger, 21 channels:
50K at 4MHz = 13ms = OK
100K at 4MHz = abort
1M at 3MHz = 333ms = OK
So it seems to be that above 3Msps it fails rapidly, even without trigger.
I'm not sure how the RLE would make any difference.
Tony

pico-coder, 2022-04-18:
Tony,
So in Rev1 the hardware trigger only worked by using a hack on the sample rate to set the "cfg_bits" as a way to get the setup to the device.  So by setting the last decimal character of the sample rate to 0x4 you got D2=0 HW trigger and 0x6 got D2=1 triggers. Did you actually do that?  I wasn't sure anybody was, and in many cases I got unexplained false triggers.
If you didn't do that then you never had HW triggers so you shouldn't miss them :).  If you did have that then maybe I was too cavalier in tossing a useful feature...Let me know...  Since you can build your own PICO uf2 IIRC it would be easy to add a fix HW trigger config back in.  The one problem with the HW triggers was that they didn't support any precapture, and they only provided value above SW triggers when the number of samples fit into RAM. If the requests samples didn't fit in RAM then you still have USB bandwidth limitations.

As for your results, you might want to benchmark your system.  Since analog channels never use RLE we can use them plus maybe some digital channels to test bandwidth.  Each analog sample is 1 Byte, and each 7 bits of digital is another byte.  Add those together and multiply by sample rate to get the USB channel bandwidth, assuming that you specified a number of samples greater than the depth of the RP2040's 200KB ram.  500k will always ensure that, but 1M is what I used to confirm the channel to be reliable or not.  Note, the values of the channels don't matter so it doesn't matter if they are connected.

Also, make sure you remove as many USB devices as possible, and make sure the PICO is directly connected to your fastest USB port (sometimes a given manufacturer will have slow ports and fast ports), and make sure you aren't using any USB hubs.

On my 10yr old Sandy bridge CPU with 3 analog and 21 digital, I can reach 125Ksps at 500k samples without aborts.  3 analog=3 bytes, 21 digital=3 bytes, so 125*(3+3)=750KB/sec.  The next available rate in rev2 is 150ksps which would be 900KB/sec but it aborts.

Copying your results with comments embedded:

With all 21 dig channels enabled, although only D2 fed with some pulses, and trigger on D2.
<Shawn> 21 digital channels take 3 bytes on the wire across USB, and take 4B of PICO RAM storage.

1M samples at 2MHz = 500ms acq time = OK
<Shawn> 1M samples would take 4MB of PICO RAM storage so you are definitely streaming here.  Without RLE this would take 3MB/sec of wire bandwidth so RLE is a win here!

200K at 2.5MHz = 80ms = OK
<Shawn> 200K would take 800KB of RAM, so again you are streaming here and RLE is working because without it you would need 3*2.5Mhz or 7.5MB/sec.

1M at 3MHz = 333ms = OK
<Shawn> And same here, RLE wins again.

100K at 4MHz = 25ms = abort
<Shawn> 100K samples is 400KB of storage, so you aren't fitting into RAM, but your higher sample rate is failing. I have found that 4-6Mhz is about the limit of what can be decoded even with high RLE rates with that many channels.
I have however found that if I drive sysclk above the spec'd 120Mhz frequency I can decode higher RLE rates correctly.  Since you seem to be right at the edge it might be worth a shot.   If you go to https://github.com/pico-coder/sigrok-pico/blob/main/pico_sdk_sigrok/sr_device.h and uncomment these two lines it will give you a 240Mhz sysclk.
It does violate your warranty, but even though I've ran a device into "failure" and then brought it back there doesn't seem to be any damage done.  If it works let me know and maybe I can make it an optional uf2 build that I release.
//#define SYS_CLK_BOOST_EN 1
//#define SYS_CLK_BOOST_FREQ 240000

With no trigger and 8 ch enabled:
<Shawn> 8 channels take 1B of DRAM storage but 2 bytes on the wire because of the funky 7 bit encoding scheme I use.  Going to 7 channels would take 1B of DRAM and 1B of wire bandwidth.
200K at 4MHz = 50ms = OK
<Shawn> 200K samples will take 200KB of storage and thus you are fitting into device RAM.
500K at 4MHz = 125ms = abort
<Shawn> So it's kind of a bummer that dropping from 21 channels to 8 channels doesn't seem to give you a higher sample rate. You'd think that if your input patterns were the same and that 3Mhz pass on 21 channel that 8 channel would would work at 4Mhz.  This might be a processing limitation where the 240Mhz build might help.  Or it could be a limitation of your host. Doing the benchmarking above, or just trying the 240Mhz build might help.
200K at 120MHz = 2ms = OK
<Obviously you are running in RAM here :) >

No trigger, 21 channels:
50K at 4MHz = 13ms = OK
<Shawn> 21 channels are 4B of memory so you are fitting in device RAM.
100K at 4MHz = abort
<Shawn> you aren't in RAM here.  Since you also aborted with triggers (above) this means that your host doing the triggers isn't slowing down processing.  I've actually seen on my raspberrypi3 that if I enable triggers at higher sample rates the triggering logic slows down the pulling of data from USB leading to aborts.
1M at 3MHz = 333ms = OK
<Shawn> Again, this works the same way as without triggers, which is expected.

pico-coder, 2022-04-18:
BTW, thanks a lot for providing information on all this.  I always kind of wonder who is using all this, so "No news" isn't always "good news" :).

tony1tf, 2022-04-18:
Hi Shawn

Thanks for looking into what is happening here. I should be able to get
away with running at 3 MHz for our application on EDSAC which has a 500kHz
clock - that should give sufficient resolution to check on the delays we
need to examine. I'll have a play with higher clock speeds in a couple of
weeks or so - got to go do other stuff now. With the present logic
analyser, I run at 4MHz and sometimes even 8MHz for better resolution, and
it is always triggered off one of the channels.
Thanks for all the work you are putting in to this. We have guys from R Pi
coming to see us in May, so I'll be showing them your work.
Currently I have a box with 8 attenuators for digital and 3 analogue divide
by 10 channels. Need to make some more attenuators and long input leads
with clips on for the remaining 13 channels. We have to pick up signals
over 3 X 30 inch racks so need long leads.

Tony

pico-coder, 2022-04-19:
thanks for the info.  But when you say triggered which one are you using:
1) SW based triggers where you left click the signal in pulseview and select level/edge etc.
2) The hack I had in rev1 (and removed in rev2) to encode trigger configs in the sample rate sent to the device?
I assume 1)...

tony1tf, 2022-04-19:
Hi Shawn

I was only talking about the Fx2 based LA compared with Pico V2. I have not
done any serious trigger tests on your vers1, so I know nothing about
hardware versus software trigger versions.
Tony

tony1tf, 2022-05-15:
Hi Shawn

I've built up my logic analyser with attenuators for use with EDSAC. Attached photo shows it, using a Maker Pi Pico which has nice blue LEDS which are good for setting the pot levels. Trouble is, the code no longer works above 8 channels. As soon as I enable more than 8 even at low speed, it crashes.  I attach a screen dump from the logging window.
I'm a bit lost!
Tony
[image]
[image]

tony1tf, 2022-05-15:
Hi Shawn

I found the problem - channels D2 to D9 have to be enabled in order to use D10 onwards.

pico-coder, 2022-06-27:
Closing this out as I think the main issues have been addressed.  Feel free to post any new issues as needed.
And thanks for the cool pics!

closed 2022-06-27 by pico-coder