pico-coder/sigrok-pico #16: Auto find of Pico

tony1tf, 2022-04-18:
Hi pico-coder

My other logic analyser using the FX2 and fx2lafw driver automatically picks up the device when clicking the PulseView icon.
Is there a reason why the search doesn't find the Pico and I can only manually search for the the Pico and serial interface?

Tony

pico-coder, 2022-04-18:
In both windows and linux (and likely Mac and others but I can't test because I don't own one) you can specify the device and parameters to the executable.
In Linux the command line would be something like this, but make sure ACM0 is replaced with your COM port.
<path>/pulseview -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 -l 2 

In windows, you need to create a short cut and then modify it.
When you install pulseview there should be a shortcut to pulseview (You can never predict where it will show up, but either under the install directory, or Users/User/AppData/Roaming/Microsoft/Windows/Start Menu/Programs/sigrok/PulsView, who knows).  But, once you find one, make a copy of it, or just create a new one from scratch (lots of links online to do that).

Then right click on the short cut, then go to properties, and under the Target you want something like this:
"C:\Program Files (x86)\sigrok\PulseView\pulseview.exe" -d raspberrypi-pico:conn=COM7:serialcomm=115200/flow=0 -l 2
(Make sure you get the right COM port number).

This would be easier if I converted to USB TMC rather than USB CDC/Serial, but that's kind of a heavy lift for me with fishing season coming up :).

Once you are connected you can save and restore the session configs, but the session configs only save channel configs, not the device configs.

I should probably get this into the user guide too, someday...

tony1tf, 2022-04-19:
I can see that this could connect to a specific device, but the old versions show a pop up searching through drivers looking for any compatible device.

pico-coder, 2022-04-19:
That search is likely based on the drivers for each USB device providing the VID/PIDs etc and then pulseview scans through a list provided by the OS.
That won't work for CDC Serial implementations that I'm currently using.
It's interesting that even though I specify the device and COM port, Pulseview still seems to do a scan of USB anyway.  Not sure who would win if you had some kind of supported USB device plugged in along with a sigrok-pico device and you specified the sigrok-pico device on the command line.  I can't test that because I don't have any other such devices....

pico-coder, 2022-06-27:
Closing.  I think either specifying the serial port on a linux command line or specifying it in a windows icon seems to make things "plug and play" to a limited extent.  Not to mention it's a lot more work on my end to implement a USB stack than it is for folks to specify a COM port, and windows USB likely requires messing with Zadig which isn't every fun either...

closed 2022-06-27 by pico-coder