pico-coder/sigrok-pico #17: Max voltage for readings

hexad, 2022-05-07:
You suggest that you use resistors to not fry the Pico, in case more than 3.3V come to contact with the board...
This means that the Pico as a logic analyzer can read signals up to 3.3 Volts?

I noticed that you have a PCB provided as well. Is the PCB just to protect the Pico, or somehow more that 3.3 voltages can be read?

One final thing, the pico can read digital channels D2-D22. This is a noob question (i just bought a pico for this project) but i don't understand what actual pins are represented...

I use this image as reference (https://www.theengineeringprojects.com/wp-content/uploads/2021/03/what-is-raspberry-pi-pico.png) and between Pico pins 2-22. there are GND pins as well. So what pins can i use for digital readings?

Thanks a lot!

hexad, 2022-05-09:
Does anyone have info on whether the pico can read voltages larger than 3.3V?

pico-coder, 2022-05-09:
The rp2040 requires the ADC pins to be less than IOVDD, and the PICO board sets IOVDD to 3.3V, so the ADC pins must be 0 to 3.3V.  Digital pins must also go between 0 and 3.3V.

The resistors are mostly recommended to protect against overvoltage or wrongly connecting jumpers -i.e. shorting Vcc to ground or something.  But, I'll be honest that I rarely use them myself :).

The PCB design was provided by one of the users of this project for his uses.  I've done all my connections via breadboarding and haven't had the need to make a board.

There is no reason that you can't add an additional external resistor divider to measure higher voltages, or add an opamp to provide voltage offsets to measure AC signals, or level shifters or all kinds of other things.  However, if you do add such things the device won't be aware of the change and the analog voltages reported by pulseview won't be correct.  But you can either use the Math function of pulseview, or edit the PICO source code to modify the offsets. See https://github.com/pico-coder/sigrok-pico/issues/14 for how tony did it.

D2-D22 map to GP2-GP22 on the pinout and ADC0,1,2 are the A0,A1,A2 pins.

The datasheet is here:
https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

hexad, 2022-05-09:
Thank you very much for your answer!
I will try to actually measure a signal tomorrow. In case i encounter errors i will report here, or else i will close the issue.
I hope everything goes well! Thank you so much for this project!

hexad, 2022-05-13:
Using the Windows program, i select `Connect to a Device`. Then i select `RaspberryPI PICO`. I select the COM port (only one is availble, so i guess it sees that something is connected there - as i have no other devices connected). Then i press `Scan for Devices` using driver above, but it sees nothing. I have tried with all diferent baud rate combinations. Does pulseview require the Pico SDK to be installed?

pico-coder, 2022-05-13:
Did you program the uf2 file to the pico device?
https://github.com/pico-coder/sigrok-pico/blob/main/pico_sdk_sigrok/build/pico_sdk_sigrok.uf2

Note that because we use USB Serial CDC the baud rate doesn't matter.

hexad, 2022-05-14:
Yes, i programed the u2f under linux (where i have the Pico SDK installed as well). So i don't select a serial port but USB right?

pico-coder, 2022-05-14:
No, you should select a serial port.  While it is a USB port, this uses the CDC Serial device version of USB so that it looks like a COM port.
You might try running pulseview in debug level 5 and see what errors if any are printed.

hexad, 2022-05-14:
Thank you very much! For some reason it worked and now workd consistently. I don't know what was to blame when it was working. However, when i press run, it starts recording only for about half a second and then the recording of the signals always stops. Do you know what causes this?
[image]

hexad, 2022-05-15:
Well, this is my first time using a Logic Analyzer, so i made obvious mistakes. It stops because that's what i get with 1k samples for 10 Hz. So everything works fine.

I want to thank you both for your amazing work and your willingness to help! I wish you all the best! Thank you very much!

closed 2022-05-15 by hexad