pico-coder/sigrok-pico #18: Consider overclockaking...?

Xiran64, 2022-05-30:
Thx for sharing to us, your excellent project! I read that pico can be overclocked, and i want to know, if you want to try to make that. =), maybe it was inestable at 400 mhz =(, but maybe....? We can load into flash?

pico-coder, 2022-06-10:
Thanks for the comment.  I actually have tried overclocking and have found the claims of clocks at 300Mhz or higher to only represent very limited usages.  
I do support overclocking the builds, not to enable a higher maximum sample rate, but to support the ability to process lower frequency sample rates to maintain peak performance across USB.
In experiments I have done I have seen failures at 288Mhz.  To support common divisors between analog and digital clocks I require a 24Mhz increment in PLL frequencies so dropped down 48Mhz to 240 for my overclocking.
To enable faster clocks uncomment this code in sr_device.h
//Frequency must be a 24Mhz multiple and less than 300Mhz to avoid known issues
//The authors PICO failed at 288Mhz, but testing with 240Mhz seemed reliable
//#define SYS_CLK_BOOST_EN 1
//#define SYS_CLK_BOOST_FREQ 240000

Xiran64, 2022-06-17:
Thx for answer pico-coder! It helps if only uses digital pins? I know the point it's to have both, =). Using the flash to store more data, will be slower? One core doing capture, send to flash, and the other sending over the usb?

pico-coder, 2022-06-27:
My testing of the attached flash showed that the maximum storage bandwidth to flash was only 1MB/sec which is right around the streaming rate of USB CDC so flash doesn't yield any additional max sample rates. 
Also, dynamically running code and writing flash yields all sorts of system stability issues.

closed 2022-06-27 by pico-coder

Xiran64, 2022-06-29:
Thx @pico-coder!!! And using an external sram?