pico-coder/sigrok-pico #19: pulseview-0.5.0-git-7e5c839-installer_rev2 shows errror on start on Windows 10

latkinso42, 2022-06-27:
Hi,
Trying desperately hard to get this to work.
I get:
pulseview application error
The application was unable to start correctly (0xc000007b).
I tried the static 64bit version 0.4.2, but it cannot see the raspberry-pi pico device.
I have upgraded the WINusb for CDC using Zadig.
Please Help. Lee@MeanStrideTech.com or latkinso42@gmail.com

pico-coder, 2022-06-27:
1) Have you tried the microsoft visual C install mentioned under 
https://sigrok.org/wiki/Windows ?

2) I don't think Zadig should be needed as the sigrok-pico just exposes a USB CDC serial port, not a regular CDC device. (I did not need to do Zadig myself when doing pico development).  AFIK, Zadig is only needed when you have a USB device whose vid/pid doesn't map to a standard released driver that windows can find, but the standard CDC serial driver in windows should be fine.  When you ran Zadig, what driver did you map it to (note that I don't think you would want to map it to pulseview, so if possiblye you might undo that mapping).

3) You might make sure Putty or Terraterm or other device can communicate with it.  If you send the letter "i" followed by a return you should get an identification string back.

4) You might try to set verbosity to 5 when starting - create a windows icon  with a path like this:
(Make sure the path is correct).
"C:\Program Files (x86)\sigrok\PulseView\pulseview.exe" -l 5"

5) You might also try to specifically target the device like this:
  (Make sure the COM port is correct, the baud rate is a don't care)
"C:\Program Files (x86)\sigrok\PulseView\pulseview.exe" -d raspberrypi-pico:conn=COM7:serialcomm=115200/flow=0 -l 5"

6) You might try other suggestions proposed by searching for "pulseview <whatever error you are seeing>".

pico-coder, 2022-08-20:
I'm closing since I haven't seen any responses for a while, hopefully the suggestions above might help others.

closed 2022-08-20 by pico-coder