pico-coder/sigrok-pico #21: PicoW

tony1tf, 2022-07-04:
With the new PicoW release last week, this seems like an ideal time to upgrade this logic analyser. Since the WiFi chip uses SPI for communication, I guess it is going to be faster than the USB serial port. Any thoughts about this Shawn?

pico-coder, 2022-07-08:
FWIW, I put in a backorder for a PICO W to spark fun, no ETA provided, and they no longer even allow backorders.  
Note that I'm not sure I want to go through the efforts of creating a sigrok wifi driver, so at best this might have the PICO W export a webpage to download a file.  But first need to prove we can actually get a noticeable bw increase over USB....

tony1tf, 2022-07-08:
Hi Shawn

One of my colleagues at EDSAC is working on a WiFi version of a Pico logic
analyser. His design is very different from yours, and is considerably more
complex, using external RAM chips, but at least I might be able to get the
WiFi code and graft it into your design.
I got my order in to Pimoroni on the day it was announced, but I guess
stocks emptied very quickly.

Tony

AapoTahkola, 2022-07-21:
You might have better luck getting an pico w from local sellers or small business. I would guess it is just temporary because there are shops with over 100 in stock here EU or nordic contries. Did the calculations and got 3 for 4.5e piece. I guess I will be using them mostly for home assistant stuff.

Anyway, I have played with esp32 wifi stuff for a long time and I can tell from experience that if pico w is anything like esp32 this stuff is fairly easy. It takes like 20 lines of code to open a port and start doing some stuff over the wifi. As this is a new chip though I might wait a bit and look at what is happening from a distance until most of the bugs are dealt with.

mtk11, 2022-10-18:
An interesting ESP32 based project - [esp32_sigrok](https://github.com/Ebiroll/esp32_sigrok) which uses Rigol emulation to communicate via WiFi with pulseview. Also would it be possible to add support for external high speed ADC/s ? As well as use the overcloking of mcu for faster sampling ?