pico-coder/sigrok-pico #23: Building Pulseview for the Raspberry Pi

kentindell, 2022-08-01:
Since I have been down the same path of building Pulseview from scratch, I thought I'd point people at this:

https://kentindell.github.io/2021/06/28/pulseview-raspberry-pi/

AapoTahkola, 2022-08-05:
You should also build libserialport from git - other versions can cause problems. Looks more detailed than other build instructions I have seen.

pico-coder, 2022-08-20:
Thanks for the comments.
I think if I had to recommend anything it would be to follow the release build information that is part of the release process.  I did find your web page when doing initial builds, but generally found that building was a iterative approach to compiling to a failure, debugging which library was missing/broking, fixing it, and repeating.  I've built so many other repos of various types I've got all kinds of weird versions of just about every library on my RPI.
Also, much of the frustration was in the addition of new devices to the build flows which require the sigrok util flows and git updates etc.  It's all kind of documented but in lots of different places.
Someday I need to get linux installer builds working. Hopefully that will work on an RPI.

rgrr, 2022-12-30:
I had problems on my Debian desktop as well with sigrok-cli and pulseview.  I think actual root of the problem had also something to do with the serial communication.
Compiling everything as pointed out in https://sigrok.org/wiki/Linux made everything work as expected.