pico-coder/sigrok-pico #24: About Sampling speed problem

ttxiaotaoge, 2022-08-17:
Hi 
When I increase the speed above 10mhz，it can not work 
log：
sr: srgn: D4 Data stream aborts with cbyte 33 char ! rdidx 268 cnt 353
sr: srgn: Ending receive on abort
sr: srgn: Sending plus to stop device stream 
pv
[image]
: Acquisition took 31.89 s

pico-coder, 2022-08-20:
thanks for the submission.
Based on the limitations here: https://github.com/pico-coder/sigrok-pico/blob/main/AnalyzerDetails.md  200k samples should never get an abort based on USB data bandwidth.

Two possibilities:
1) The abort detection logic on the pico may have a bug.
2 At 40Msps the data rate coming in is fast enough to slow down memory access to cause a PIO delay to memory.  I don't think that should be possible though since PIO should have priority to memory.

I'll need to try to recreate this on my end.  Until then, you might try:
1) Going below 200k samples
2) Reducing sample rate if that works for you

pico-coder, 2022-09-23:
I tried to reproduced this but the abort detect logic seems to be working correctly.
I'm guessing that what might be going on is related to the SPI flash/EEPROM  decoder you have running.
If that decoder enables any kind of triggering then that might put the device into a free running sample mode and in that case the USB bandwidth won't be able to keep up.

Can you run again, but this time with debug level of 4?  (A 5 level would be useful but might require a very large background buffer - if you can make that work it would be great, but a level of 4 is probably good enough).

pico-coder, 2023-03-27:
Closing as I've gotten no updates.  I'm assuming that the user somehow enabled free running sample mode and overflowed the usb link capacity

closed 2023-03-27 by pico-coder

vanbwodonk, 2023-11-07:
Got same problem with 15Mhz sampling, 10Mhz is doing fine. 
This is log output:
```
sr: log: libsigrok loglevel set to 4.
srd: libsigrokdecode loglevel set to 4.
sr: srpico: config_get given key 30000
sr: session: Creating our own main context.
sr: session: Starting.
sr: hwdriver: raspberrypi-pico: Starting acquisition.
sr: srpico: Enter acq start
sr: srpico: dsbstart 3
sr: srpico: c 0 enabled 0 name A0 
sr: srpico: Channel enable masks D 0x1FFFFF A 0x6
sr: srpico: c 1 enabled 0 name A1 
sr: srpico: Channel enable masks D 0x1FFFFF A 0x4
sr: srpico: c 2 enabled 0 name A2 
sr: srpico: Channel enable masks D 0x1FFFFF A 0x0
sr: srpico: c 0 enabled 1 name D2 
sr: srpico: Channel enable masks D 0x1FFFFF A 0x0
sr: srpico: c 1 enabled 1 name D3 
sr: srpico: Channel enable masks D 0x1FFFFF A 0x0
sr: srpico: c 2 enabled 0 name D4 
sr: srpico: Channel enable masks D 0x1FFFFB A 0x0
sr: srpico: c 3 enabled 0 name D5 
sr: srpico: Channel enable masks D 0x1FFFF3 A 0x0
sr: srpico: c 4 enabled 0 name D6 
sr: srpico: Channel enable masks D 0x1FFFE3 A 0x0
sr: srpico: c 5 enabled 0 name D7 
sr: srpico: Channel enable masks D 0x1FFFC3 A 0x0
sr: srpico: c 6 enabled 0 name D8 
sr: srpico: Channel enable masks D 0x1FFF83 A 0x0
sr: srpico: c 7 enabled 0 name D9 
sr: srpico: Channel enable masks D 0x1FFF03 A 0x0
sr: srpico: c 8 enabled 0 name D10 
sr: srpico: Channel enable masks D 0x1FFE03 A 0x0
sr: srpico: c 9 enabled 0 name D11 
sr: srpico: Channel enable masks D 0x1FFC03 A 0x0
sr: srpico: c 10 enabled 0 name D12 
sr: srpico: Channel enable masks D 0x1FF803 A 0x0
sr: srpico: c 11 enabled 0 name D13 
sr: srpico: Channel enable masks D 0x1FF003 A 0x0
sr: srpico: c 12 enabled 0 name D14 
sr: srpico: Channel enable masks D 0x1FE003 A 0x0
sr: srpico: c 13 enabled 0 name D15 
sr: srpico: Channel enable masks D 0x1FC003 A 0x0
sr: srpico: c 14 enabled 0 name D16 
sr: srpico: Channel enable masks D 0x1F8003 A 0x0
sr: srpico: c 15 enabled 0 name D17 
sr: srpico: Channel enable masks D 0x1F0003 A 0x0
sr: srpico: c 16 enabled 0 name D18 
sr: srpico: Channel enable masks D 0x1E0003 A 0x0
sr: srpico: c 17 enabled 0 name D19 
sr: srpico: Channel enable masks D 0x1C0003 A 0x0
sr: srpico: c 18 enabled 0 name D20 
sr: srpico: Channel enable masks D 0x180003 A 0x0
sr: srpico: c 19 enabled 0 name D21 
sr: srpico: Channel enable masks D 0x100003 A 0x0
sr: srpico: c 20 enabled 0 name D22 
sr: srpico: Channel enable masks D 0x3 A 0x0
sr: srpico: bps 1 
sr: srpico: At get_dev_cfg
sr: srpico: A0 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: srpico: A1 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: srpico: A2 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: srpico: Entering fixed sample mode
sr: session: bus: Received SR_DF_HEADER packet.
sr: srpico: dsbstartend 3
sr: session: bus: Received SR_DF_TRIGGER packet.
sr: srpico: rx string ��������[�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�#
sr: session: bus: Received SR_DF_LOGIC packet (24534 bytes, unitsize = 3).
[... 13 lines cut ...]
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23439 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (11295 bytes, unitsize = 3).
sr: srpico: rx string R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�R�J�!!!!!!#
sr: session: bus: Received SR_DF_LOGIC packet (26205 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23439 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23439 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: session: bus: Received SR_DF_LOGIC packet (23436 bytes, unitsize = 3).
sr: srpico: D4 Data stream aborts with cbyte 33 char ! rdidx 214 cnt 573
sr: session: bus: Received SR_DF_LOGIC packet (11949 bytes, unitsize = 3).
sr: srpico: Ending receive on abort
sr: srpico: At dev_acquisition_stop
sr: session: bus: Received SR_DF_END packet.
sr: srpico: Sending plus to stop device stream
sr: session: fd_source_finalize: key 0x7fa63c012770
sr: session: Stopped.
pv: Acquisition took 0.14 s
```