pico-coder/sigrok-pico #27: SigrokBuildNotes.md

gnbl, 2022-09-23:
https://github.com/pico-coder/sigrok-pico/blob/main/SigrokBuildNotes.md should have another bullet point `4b) apply generated patch`

gnbl, 2022-09-23:
Also, building libsigrok with raspberripi-pico generates lots of -Wformat type warnings.

gnbl, 2022-09-23:
sigrok-cli would immediately exit with "... undefined symbol: srd_session_send_eof" so I went back to https://www.sigrok.org/wiki/Linux#Building_(manually) and tried all dependencies - Python binding support was not found despite numpy being installed (Ubuntu 22.04).
After reconfiguring and rebuilding everything, the pico was detected.
Pulseview just finished building and also detected the pico - awesome!

But unfortunately, 

```
sigrok-cli -l 2 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --config samplerate=100000 --channels D2,D3,D4,D5,D6,D7,D8,D9,D10,D11,D12,D13 --samples 100k
sr: srgn: Drain reads
sr: srgn: Drain reads done
sr: srgn: sr_err level logging enabled
sr: srgn: sr_warn level logging enabled
libsigrok 0.6.0-git-e280f1f
Acquisition with 12/24 channels at 100 kHz
D2:00
D3:00
D4:00
D5:00
D6:00
D7:00
D8:00
D9:00
D10:00
D11:00
D12:00
D13:00
Device only sent 2 samples.
```

8 channels seem to work fine at 100kHz, but with more channels, only two samples are shown in Pulseview (0 and 10 us). Same at 50 kHz

```
sr: [02:05.935741] srgn: at config_get key 30000
sr: [02:05.935777] srgn: sample rate get of 100000
sr: [02:05.935787] hwdriver: sr_config_get(): key 30000 (samplerate) sdi 0x55a913503f30 cg NULL -> uint64 100000
sr: [02:05.935815] session: Creating our own main context.
sr: [02:05.935844] session: Starting.
sr: [02:05.935852] hwdriver: raspberrypi-pico: Starting acquisition.
sr: [02:05.935862] srgn: Enter acq start
sr: [02:05.935868] srgn: dsbstart 3
sr: [02:05.935878] serial: Draining serial port /dev/ttyACM0.
sr: [02:05.935912] serial: Wrote 1/1 bytes.
sr: [02:05.935921] serial: Draining serial port /dev/ttyACM0.
sr: [02:05.935952] srgn: c 0 enabled 0 name A0
sr: [02:05.935961] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:05.946038] serial: Wrote 4/4 bytes.
sr: [02:05.947545] serial: Read 1/1 bytes.
sr: [02:05.947560] srgn: c 1 enabled 0 name A1
sr: [02:05.947568] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:05.957644] serial: Wrote 4/4 bytes.
sr: [02:05.958538] serial: Read 1/1 bytes.
sr: [02:05.958547] srgn: c 2 enabled 0 name A2
sr: [02:05.958552] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:05.968630] serial: Wrote 4/4 bytes.
sr: [02:05.969582] serial: Read 1/1 bytes.
sr: [02:05.969589] srgn: c 0 enabled 1 name D2
sr: [02:05.969595] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:05.979674] serial: Wrote 4/4 bytes.
sr: [02:05.981619] serial: Read 1/1 bytes.
sr: [02:05.981631] srgn: c 1 enabled 1 name D3
sr: [02:05.981643] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:05.991727] serial: Wrote 4/4 bytes.
sr: [02:05.992667] serial: Read 1/1 bytes.
sr: [02:05.992679] srgn: c 2 enabled 1 name D4
sr: [02:05.992689] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:06.002776] serial: Wrote 4/4 bytes.
sr: [02:06.004690] serial: Read 1/1 bytes.
sr: [02:06.004709] srgn: c 3 enabled 1 name D5
sr: [02:06.004718] srgn: Channel enable masks D 0xFF A 0x0
sr: [02:06.014804] serial: Wrote 4/4 bytes.
[... 105 lines cut ...]
sr: [02:07.387395] serial: Read 4/31999 bytes.
sr: [02:07.387416] srgn: Entry wrptr 0 bytes_rem 32000 len 4
sr: [02:07.387425] srgn: rx string $4+#
sr: [02:07.387431] srgn: rx len 4 bytes_avail 4l sent_samples 2l wrptr 0
sr: [02:07.387439] srgn: Data stream stops with cbyte 36 char $ rdidx 0 sbytes 0 cnt 4
sr: [02:07.387447] srgn: process slice avail 4 rdptr 0 sb 0 byte_cnt 4
sr: [02:07.387455] srgn: Residual shift rdptr 0 wrptr 4
sr: [02:07.387462] srgn: Stopped, checking byte_cnt
sr: [02:07.387468] srgn: Byte_cnt check device cnt 4 host cnt 4
sr: [02:07.387475] srgn: ****at dev_acquisition_stop
sr: [02:07.387481] session: bus: Received SR_DF_END packet.
sr: [02:07.487625] session: fd_source_finalize: key 0x55a91305a5c0
sr: [02:07.487655] session: Stopped.
Acquisition took 1.23 s
```

Trying a different set of channels (D10-D22):

sr: [05:06.263333] srgn: Digital channel mask 0x1FFF00 not continous
sr: [05:06.263343] session: Could not start raspberrypi-pico device /dev/ttyACM0 acquisition.
sr: [05:06.263353] hwdriver: raspberrypi-pico: Stopping acquisition.
sr: [05:06.263360] srgn: ****at dev_acquisition_stop
sr: [05:06.263367] session: bus: Received SR_DF_END packet.
sr: [05:06.363499] session: Cannot remove non-existing event source 0x55a91305a5c0.
Notifying user of session error: "generic/unspecified error"

gnbl, 2022-09-23:
100k samples at 100kHz for all digital channels seems to work. 
D2-D17 fails, D2-D18+ works

pico-coder, 2022-09-23:
The protocol supports run length encoding where if you get the same sample value over and over a repeat count is sent instead of sending all the same value.
What kind of inputs are you using here?  I'm guessing that maybe all of the inputs are left unconnected?

gnbl, 2022-09-23:
Thanks for your response. And the project.
Yes, floating inputs, so random level. But I figured it has something to do with the RLE channel "boundaries" since my results appear repeatable (acquisition immediately fails or works OK).
But I will repeat test with defined levels asap.

pico-coder, 2022-09-23:
Bug found.  In the rev2 release I wrote some optimized loops that had to be cut and pasted a few times  (send_slices_1B,2B,4B) and in the 2B version what should have been a function call to check_rle ended up being an instance.
i.e. it should be 
check_rle();
but was just
check_rle;
The samples are split into chunks on the dma engine and for every chunk we send one sample to start, thus two total samples, but never sent the rle....
Working on a fix along with some other changes.

pico-coder, 2022-09-26:
The bug in rle has been fixed and released and new uf2 is available.  Please try it out and let me know how it goes.
Remaining stuff:
1) Wformat errors - yes, I have a bunch of debug related printfs which don't use the correct printf symbol to be portable across architectures.  It only shows up when you build certain sizes.  It doesn't effect functionaly AFIK so may be a while (never) before I get it cleaned up.
2) Need to have a look through the build notes and update accordingly.  What I really want to do is just get the sigrok builds going and post installers here so that folks don't have to go through the pains.

gnbl, 2022-09-27:
Great, will test asap.

1. might affect how the pull request is received "upstream"

gnbl, 2022-09-30:
First test without issue! Feel free to close. Thanks!

closed 2025-10-15 by pico-coder