pico-coder/sigrok-pico #28: Allow 12 bit ADC values (feature request)

Bentschi, 2022-12-23:
Hello,
this is more of a feature request than an issue.
First of all, i really love that project and was going a little through the code.
I would really love to have access to the full 12 bit ADC values.
As far as limitations go i think it shouldn't impact the sampling or transfer rates a lot.
I know it would require to change the driver and obviously the firmware.
I am not yet confident enough to do the PIO and DMA stuff for myself, but I would be willing to learn it.

Well, my actual questions are, can it be done? do i have to do it myself? and is there something i am not seeing and i would run straight into a wall?

Bentschi, 2022-12-24:
Hello and Merry Christmas,
I did some research on it and maybe I also didn't read AnalyzerDetails.md carefully enough.

> 3 Analog Channels
> 
> Any combinations of A0 (ADC0 pin 31),A1(ADC1 pin 32) and A2(ADC2 pin 34) can be enabled. Channels are only 7 bit accurate because: 1) Even though the ADC gives a 12 bit value, the ENOB of the RP2040 is only ~8 bits. 2) 7 bits makes an easy wire encoding that avoids ASCII characters that can be messed up by serial drivers (see SerialProtocol.md) 3) 7 bits still gives 20mV accuracy and 128 divisions which is usually plenty of separation.

The reason i was asking was because i created a circuit with 2 inverting OpAmps to map an input from -15V to +15V to an output of 0 to 3.3V for the ADC.
With 7 bit i get only accuracy down to 0.23V.
But I think I understand the reasoning behind the 7 bits now, and i am sorry if my first post sounds a bit harsh, I didn't mean that it should be integrated in this project at all cost and I kinda want to figure some stuff out for myself (but also didn't want to waste my time if it meant changing 2 lines, which it clearly does not).
So I am thinking to get the full 12 bit the ADC can provide i would have to take an entirely different approach than using DMA like in this project.

pico-coder, 2022-12-24:
Using DMA transfers for 12bit reads should work just fine as long as the DMA fifo is configured to not discard 4 bits.  You found the official documentation, but the unofficial reason for not supporting more resolution was that I ran out of time trying to get everything else working and figured I'd wait to see if anyone cared about higher resolution.  It is also looking like I might get this repo merged to mainline sigrok repo so that will take priority, but I'll keep this open and think about supporting higher resolution.  Merry Christmas.

Bentschi, 2022-12-31:
Thank you very much for clarifying that.
And again thanks for the project in general, the Pico as logic analyzer was quite helpful to me so far.

pico-coder, 2024-12-02:
ok. With my first commit getting accepted around a year ago I can't use that as an excuse anymore ;).
With the release of the RP2530 ENOB went from 8.5 to 9.5.  So in theory going to 2 7 bit "bytes" could give 1.5 extra bits, which is... maybe... interesting.. or not??? .  The libsigrok code definitely handles 1 7 bit "byte" and should handle 2,3, or 4 but I've never really tested it.  
And, overall, the RP2530 is pretty underwhelming compared to an RP2040 for the logic analyzer/ADC use case. (USB isn't faster, ADC isn't fundamentally better).  So I haven't bothered to buy one yet.
Anybody out there care enough to try to enable 12 bit mode in the PICO device code, or have a substantial use case?  Not that I'm lazy or anything, but having folks with strong motivations for features might help me consider adding it....
Note that 2x increase would effectively halve the number of samples that could be stored and/or would cut the max sample rate in half that could be issued across USB, so it wouldn't be "free".

pico-coder, 2025-03-20:
So the current relased code of sigrok does support the parsing of multi-bytes per sample analog encodings.
I'm working on various special build types to give more digital signals or rp2350 support or whatever....
but the question is, is anybody interested in 12 bit ADC values?  realistically you only reach ENOB of 9.5, and already have 7 bits in the current config.  And to get it, you have to cut your maximum streaming sample rate in half.
So I'll put it a to vote, or at least a request for feedback to see who is interested.  Based on @Bentschi 's responses I think he's happy with the resolution 7 bits provides.