pico-coder/sigrok-pico #29: Support for DSView ?

mtk11, 2023-01-10:
Have you considered supporting [DSView](https://github.com/DreamSourceLab/DSView) ? It looks better than PulseView, the oscilloscope window looks as oscilloscope and it is also based on libsigrok.

pico-coder, 2023-01-10:
While it's open source, DSView only supports DS devices, and I have a feeling that they wouldn't want to support other devices in their downloads.  While I do provide Windows executables I'm really hoping to get out of the business of supplying executables (libsigrok repo moderators are reviewing my pull request now).
Looking at the DSView screenshots it doesn't look all that different from what I see in PulseView.  Can you give a better description of what you mean by it looking like an oscilloscope?

pico-coder, 2023-01-10:
Rule #1 with Open Source anything:
- Never say it's easy unless you've posted the pull request to prove it :).

Based on a very quick scan it looks like they may have removed the usb CDC serial libraries that baseline sigrok provides.  My code assumes a "serial" like implementation.  Also, it looks like they may have extended the configuration space which we'd need to migrate to interact with the DSView code.

Even if we aren't a competitor, which we definitely aren't, that doesn't mean DSLogic will want to support what is in this repo, nor will they likely want to release builds to include this device.  You are welcome to prove me wrong on that front by submitting an issue to their github repo on the topic and see what they say.
Alternately, it might be worthwhile to see if their FFT and other features could be pulled into the mainline pulseview repo.  I haven't looked into what that might look like, but probably not easy either.....

FFT, more advanced triggering and other features are definitely interesting, it's just not clear what the easiest path to done is...

pico-coder, 2024-12-02:
DSView closed the ticket without a direct answer, so I think that means "no".  Closing this as I don't see a practical way to get RP2040 support on DSView

closed 2024-12-02 by pico-coder