pico-coder/sigrok-pico #32: Python driver?

tannewt, 2023-01-30:
Do you have a Python driver for this? Maybe one that uses PySerial? I'm looking into reimplementing sigrok-cli in Python with plugins so that hardware, file format and decoder support isn't blocked on a central repo.

tannewt, 2023-02-09:
I've written a Python driver: https://github.com/pysigrok/hardware-raspberrypi-pico/blob/main/pysigrok_raspberrypi_pico.py

I've also modified the firmware a bit to support different boards: https://github.com/pysigrok/sigrok-pico

pico-coder, 2023-02-10:
How do I run it?  All of the python I've written are standalone python files I run from Spyder Python windows so I'm not very savy on this kind of thing.
What do you use to plot the results?

tannewt, 2023-02-10:
I just tested on windows. Usually I’m on Linux or Mac. To run it I installed python by typing ‘python’ into power shell. Then I ‘pip install pysigrok-hardware-raspberrypi-pico’.

There terse instructions here: https://github.com/pysigrok/pysigrok#quick-start

‘pysigrok-cli’ is meant to be compatible with ‘sigrok-cli’. So, it can write a .sr file you can load into pulseview.

We made a #pysigrok channel on the adafruit discord if you need help. https://adafru.it/discord

pico-coder, 2024-02-05:
I'm way behind in updates.  Since you have 21 stars I'll assume what you have is useful.  Rather than trying to merge what you have in (I'm nearly functionally illiterate in github) I think I may just add a link to your repo at my intro page, and close this out?