pico-coder/sigrok-pico #33: pulseview not connecting to my pico w

chrisaccount, 2023-02-07:
i installed pulseview from pulseview-0.5.0-git-7e5c839-installer_rev2.exe on my windows 10 laptop
i compiled pico code in my pico w envronment.

in pulseview i select device "RaspberryPI PICO (raspberryp-pico)", COM3, baud 115200, then "Scan for devices using driver above".  Nothing happens.

then i close pulseview and open putty to COM3 115200.  type "i" return and i get "SRPICO,A031D21,02".

Any thoughts on what im doing wrong would be appreciated.  Or what i can do to help you understand my problem.

thanks
chris

pico-coder, 2023-02-07:
Try rebooting.  There is some underlying issues with the serial library
that seems to prevent the device from being accessed.  If a reboot fails,
then get a level 4 log report and post it here.

chrisaccount, 2023-02-07:
Thanks so much for reply.  i just tried reboot and not luck.  later today i will read a bit more to find out how to get a "level 4 log report" and post it here.

pico-coder, 2023-02-07:
I apologize for what I'm about to say, but this usually comes down to plugging/replugging/rebooting/restarting until it works.
Part of it may have to do with the serial port being unintentionally grabbed by another program so make sure you have no other serial apps running.
My general impression is that putty (and minicom on linux) seem to have a better mechanism to open serial ports that the serial library used by pulseview doesn't have.  I've never been able to dive deep into the issue as it tends to resolve itself over time.

chrisaccount, 2023-02-07:
thanks so much.  it is working now !!  tried connecting/disconnecting/launch/close/...  a bunch.  never seemed to do anything after  selecting "Scan for devices using driver above".  however then i noticed that in the device select pull down it had options "Connect to device", "Demo Device", and "Raspberry PI PICO" as options.  Could be that it was there all the time.  Selecting "Raspberry PI PICO" worked.  So i would say this was an operator error.

Thanks so much.  love being able to use pico as logic analyzer cause the $10 (24Mhz) ones from amazon do not have analog :-)

chrisaccount, 2023-02-07:
closing this. cause it was operator error on my part.

closed 2023-02-07 by chrisaccount

chrisaccount, 2023-02-07:
p.s.  for me i can connect to pico ok if i first open putty, connect to com3 at 1115200, type i[enter] several times (7 ish), close putty.  then open pulseview and it it connects OK to pico on first try.  3 out of 3 times i tried it.

hcirnieh, 2023-07-08:
SOLUTION: Do not specify a baud rate in pulseview, then search for devices