pico-coder/sigrok-pico #34: regarding channel numbering.

hanetzer, 2023-03-03:
You mention 21 channels, D2-D22, and as being the markings on the pcb.
However, my picos and the upstream documentation have the gpio pins
numbered as GPn, not Dn. I assume thats a 1:1 match?

pico-coder, 2023-03-06:
The numbering is per the PICO board signal names, not the RP2040 pin names.
https://datasheets.raspberrypi.com/pico/Pico-R3-A4-Pinout.pdf
Thus D2-D22 are GP2 to GP22, and A0-A3 are ADC0-ADC3.
I do need to clarify that in the AnalyzerDetails.md.  Also, I'm in the process of supporting more descriptive channel names.

hanetzer, 2023-03-07:
Nice, that's what I figured but always good to verify. Unfortunately the pico can't work fast enough
as a logic analyzer for my purposes, but I'm gonna try to adapt the clocked_input example from
pico-examples to sample the lpc clock at 33.3Mhz, do some light parsing to filter out empty cycles,
and forward them over usb (lpc is 33.3Mhz and about a byte of data each cycle, which is too fast
for the 12Mbps the pico's usb is capable of)

pico-coder, 2023-03-07:
This implementation supports run length encoding (RLE) over the wire so that duplicate samples are not sent.  So there is a chance it might work, though at 33Msps the Pico cpu may not be able to process it fast enough.

pico-coder, 2024-12-02:
Mostly this issue was about pin naming and assignments which is cleared up documentation.
https://github.com/sigrokproject/libsigrok/pull/256 is inflight to support better channel rename support, but that is covered in other issues so closing this one.

closed 2024-12-02 by pico-coder