pico-coder/sigrok-pico #35: Attempting to use a trigger above 10 MHz doesn't work

wrongsyntax, 2023-04-16:
Attempting to set the sample speed to any value above 10 MHz doesn't work, for any number of samples.

PulseView starts to run the acquisition but immediately stops, with no output.

<details>
<summary>Level 4 output</summary>

```py
sr: [03:17.970712] srpico: at config_get key 30000
sr: [03:17.970813] session: Using thread-default main context.
sr: [03:17.970840] session: Starting.
sr: [03:17.970858] hwdriver: raspberrypi-pico: Starting acquisition.
sr: [03:17.970940] srpico: Enter acq start
sr: [03:17.970967] srpico: dsbstart 3
sr: [03:17.971208] srpico: c 0 enabled 0 name A0
sr: [03:17.972339] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:17.972431] srpico: swack drops 2 bytes 36 51
sr: [03:17.972473] srpico: swack drops 2 bytes 53 51
sr: [03:17.972505] srpico: swack drops 2 bytes 43 10
sr: [03:17.984313] srpico: c 1 enabled 0 name A1
sr: [03:17.984351] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:17.996444] srpico: c 2 enabled 0 name A2
sr: [03:17.996518] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.009365] srpico: c 0 enabled 1 name D2
sr: [03:18.009405] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.022592] srpico: c 1 enabled 0 name D3
sr: [03:18.022807] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.035525] srpico: c 2 enabled 0 name D4
sr: [03:18.035567] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.049718] srpico: c 3 enabled 0 name D5
sr: [03:18.049787] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.062541] srpico: c 4 enabled 0 name D6
sr: [03:18.062581] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.074683] srpico: c 5 enabled 0 name D7
sr: [03:18.074718] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.086615] srpico: c 6 enabled 0 name D8
sr: [03:18.087369] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.098636] srpico: c 7 enabled 0 name D9
sr: [03:18.098673] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.110697] srpico: c 8 enabled 0 name D10
sr: [03:18.110752] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.123772] srpico: c 9 enabled 0 name D11
sr: [03:18.123829] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.134740] srpico: c 10 enabled 0 name D12
sr: [03:18.134782] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.146914] srpico: c 11 enabled 0 name D13
sr: [03:18.146955] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.159822] srpico: c 12 enabled 0 name D14
sr: [03:18.159865] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.172863] srpico: c 13 enabled 0 name D15
sr: [03:18.172901] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.184912] srpico: c 14 enabled 0 name D16
sr: [03:18.184939] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.197917] srpico: c 15 enabled 0 name D17
sr: [03:18.197953] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.210101] srpico: c 16 enabled 0 name D18
sr: [03:18.210161] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.222081] srpico: c 17 enabled 0 name D19
sr: [03:18.222139] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.234021] srpico: c 18 enabled 0 name D20
sr: [03:18.234061] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.247270] srpico: c 19 enabled 0 name D21
sr: [03:18.247332] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.259171] srpico: c 20 enabled 0 name D22
sr: [03:18.259215] srpico: Channel enable masks D 0x1 A 0x0
sr: [03:18.271143] srpico: bps 1
sr: [03:18.284237] srpico: At get_dev_cfg
sr: [03:18.297144] srpico: A0 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: [03:18.308446] srpico: A1 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
sr: [03:18.322220] srpico: A2 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#
[... 792 lines cut ...]
sr: [07:36.153172] srpico: rx string ��[...]��#
sr: [07:36.290465] session: bus: Received SR_DF_LOGIC packet (22560 bytes, unitsize = 3).
sr: [07:36.290694] session: bus: Received SR_DF_ANALOG packet (7520 samples).
sr: [07:36.291385] srpico: rx string ��[...]��#
sr: [07:36.528878] session: bus: Received SR_DF_LOGIC packet (25830 bytes, unitsize = 3).
sr: [07:36.529186] session: bus: Received SR_DF_ANALOG packet (8610 samples).
sr: [07:36.529389] srpico: Ending: sent 20000 of limit 20000 samples byte_cnt 54779
sr: [07:36.529974] srpico: rx string ��[...]��#
sr: [07:36.754054] srpico: Ending: sent 20000 of limit 20000 samples byte_cnt 86777
sr: [07:36.764211] srpico: rx string ��[...]��!!!!!!!!!#
sr: [07:36.921841] srpico: Data stream aborts with cbyte 33 char ! rdidx 0 sbytes 22792 cnt 86777
sr: [07:36.922547] srpico: Ending receive on abort
sr: [07:36.922565] srpico: ****at dev_acquisition_stop
sr: [07:36.922578] session: bus: Received SR_DF_END packet.
sr: [07:36.922619] srpico: Sending plus to stop device stream
sr: [07:37.026239] session: fd_source_finalize: key 0x60000302fc50
sr: [07:37.026369] session: Stopped.
Acquisition took 0.88 s
```

</details>

I can provide more logs in other scenarios as needed as well.

([...] is used where the same character repeats)

pico-coder, 2023-04-16:
The abort that is sent means the USB link wasn't able to keep up with the
data rate which is expected at 10msps.

On Sat, Apr 15, 2023, 6:44 PM Uzair ***@***.***> wrote:

> Attempting to set the sample speed to any value above 10 MHz doesn't work,
> for any number of samples.
>
> PulseView starts to run the acquisition but immediately stops, with no
> output.
> Level 4 output
>
> sr: [03:17.970712] srpico: at config_get key 30000sr: [03:17.970813] session: Using thread-default main context.sr: [03:17.970840] session: Starting.sr: [03:17.970858] hwdriver: raspberrypi-pico: Starting acquisition.sr: [03:17.970940] srpico: Enter acq startsr: [03:17.970967] srpico: dsbstart 3sr: [03:17.971208] srpico: c 0 enabled 0 name A0sr: [03:17.972339] srpico: Channel enable masks D 0x1 A 0x0sr: [03:17.972431] srpico: swack drops 2 bytes 36 51sr: [03:17.972473] srpico: swack drops 2 bytes 53 51sr: [03:17.972505] srpico: swack drops 2 bytes 43 10sr: [03:17.984313] srpico: c 1 enabled 0 name A1sr: [03:17.984351] srpico: Channel enable masks D 0x1 A 0x0sr: [03:17.996444] srpico: c 2 enabled 0 name A2sr: [03:17.996518] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.009365] srpico: c 0 enabled 1 name D2sr: [03:18.009405] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.022592] srpico: c 1 enabled 0 name D3sr: [03:18.022807] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.035525] srpico: c 2 enabled 0 name D4sr: [03:18.035567] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.049718] srpico: c 3 enabled 0 name D5sr: [03:18.049787] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.062541] srpico: c 4 enabled 0 name D6sr: [03:18.062581] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.074683] srpico: c 5 enabled 0 name D7sr: [03:18.074718] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.086615] srpico: c 6 enabled 0 name D8sr: [03:18.087369] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.098636] srpico: c 7 enabled 0 name D9sr: [03:18.098673] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.110697] srpico: c 8 enabled 0 name D10sr: [03:18.110752] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.123772] srpico: c 9 enabled 0 name D11sr: [03:18.123829] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.134740] srpico: c 10 enabled 0 name D12sr: [03:18.134782] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.146914] srpico: c 11 enabled 0 name D13sr: [03:18.146955] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.159822] srpico: c 12 enabled 0 name D14sr: [03:18.159865] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.172863] srpico: c 13 enabled 0 name D15sr: [03:18.172901] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.184912] srpico: c 14 enabled 0 name D16sr: [03:18.184939] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.197917] srpico: c 15 enabled 0 name D17sr: [03:18.197953] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.210101] srpico: c 16 enabled 0 name D18sr: [03:18.210161] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.222081] srpico: c 17 enabled 0 name D19sr: [03:18.222139] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.234021] srpico: c 18 enabled 0 name D20sr: [03:18.234061] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.247270] srpico: c 19 enabled 0 name D21sr: [03:18.247332] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.259171] srpico: c 20 enabled 0 name D22sr: [03:18.259215] srpico: Channel enable masks D 0x1 A 0x0sr: [03:18.271143] srpico: bps 1sr: [03:18.284237] srpico: At get_dev_cfgsr: [03:18.297144] srpico: A0 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#sr: [03:18.308446] srpico: A1 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#sr: [03:18.322220] srpico: A2 scale 0.025700 offset 0.000000 response #25700x0# tokens #25700# #0#sr: [03:18.347108] srpico: Trigger value idx 0 match 3sr: [03:18.372924] srpico: Allocating pretrig buffers size 20000sr: [03:18.372963] srpico: Entering sw triggered modesr: [03:18.373007] session: bus: Received SR_DF_HEADER packet.sr: [03:18.373027] srpico: dsbstartend 3sr: [03:18.391235] srpico: rx string ����������[...]��#sr: [03:18.413865] srpico: rx string ��[...]��f�!!!!!!#sr: [03:18.415779] srpico: D4 Data stream aborts with cbyte 33 char ! rdidx 88 cnt 353sr: [03:18.415858] srpico: Ending receive on abortsr: [03:18.415873] srpico: ****at dev_acquisition_stopsr: [03:18.415883] session: bus: Received SR_DF_END packet.sr: [03:18.415912] srpico: Sending plus to stop device streamsr: [03:18.516101] session: fd_source_finalize: key 0x600003f8b980sr: [03:18.516163] session: Stopped.
>
> Level 5 output
>
> sr: [00:51.132975] srpico: at config_get key 30000sr: [00:51.133610] srpico: sample rate get of 15000000sr: [00:51.133661] hwdriver: sr_config_get(): key 30000 (samplerate) sdi 0x600003d39260 cg NULL -> uint64 15000000sr: [00:51.133955] session: Checking trigger:sr: [00:51.133996] session: Stage 0 match on channel D2, match 3sr: [00:51.134169] session: Using thread-default main context.sr: [00:51.134198] session: Starting.sr: [00:51.134222] hwdriver: raspberrypi-pico: Starting acquisition.sr: [00:51.134246] srpico: Enter acq startsr: [00:51.134266] srpico: dsbstart 3sr: [00:51.134294] serial: Draining serial port /dev/tty.usbmodem14201.sr: [00:51.134570] serial: Wrote 1/1 bytes.sr: [00:51.134620] serial: Draining serial port /dev/tty.usbmodem14201.sr: [00:51.136932] srpico: c 0 enabled 0 name A0sr: [00:51.136990] srpico: Channel enable masks D 0x1 A 0x0sr: [00:51.137035] ser [... cut ...] : [02:33.574691] serial: Wrote 5/5 bytes.sr: [02:33.575421] serial: Read 1/1 bytes.sr: [02:33.575459] srpico: c 11 enabled 0 name D13sr: [02:33.575472] srpico: Channel enable masks D 0x1 A 0x0sr: [02:33.587318] serial: Wrote 5/5 bytes.sr: [02:33.588403] serial: Read 1/1 bytes.sr: [02:33.588461] srpico: c 12 enabled 0 name D14sr: [02:33.588487] srpico: Channel enable masks D 0x1 A 0x0sr: [02:33.601187] serial: Wrote 5/5 bytes.sr: [02:33.602540] serial: Read 1/1 bytes.sr: [02:33.602593] srpico
>
>

wrongsyntax, 2023-04-16:
Does that mean it's not possible to use triggers past 10 MHz? It works as expected up to 120 MHz without any triggers.

pico-coder, 2023-04-18:
Practically, yes.  
https://github.com/pico-coder/sigrok-pico/blob/main/AnalyzerDetails.md has more details.
https://github.com/pico-coder/sigrok-pico/issues/22 has a reference to another pico implementation.  It has hardware triggering up to 100Mhz or so, but is limited to never capture more than about 50k samples.  Merging the two types has been on my TODO list, but what time I have has been spent on trying to get this merged into the sigrok mainline.

dovandung, 2025-04-08:
Could you update the code when in trigger mode capture , if the PICO buffer full but still has no trigger, we just rearm it again. Because now it stop and need to press "Run" again

pico-coder, 2025-10-15:
No progress here as I've been working on other changes.  But I at least wanted to update all the open issues with a status.  The one problem with this proposal is that the CPU core on PICO may not be fast enough to actually process the samples looking for a trigger and thus we might miss a trigger as it goes by.  Depending on the use case, that may be ok as long as the event repeats so that you eventually get a sample with a trigger.  But, in general this is really pushing the limits of what the PICOs are really capable of, and combined limitations on how pulseview defines triggering I'm not sure there is a general purpose answer.