pico-coder/sigrok-pico #36: Windows Pulseview cant work, it shows data fragmently and much shorter than waht I want.

taorye, 2023-04-20:
I have download the version you released, but I cant get it work.

It just shows the data about 500ms long and then it stopped sampling even if I set 1M samples(sr:200K) as it should sample last for 5s.

I run it on my Windows 11, and then I test another LA device like fx2la. It occurs again.

I also download the official(v0.4.2) version, and it works, keep showing as its sampling.

So can you tell me how can I get it work normally?

taorye, 2023-04-20:
Like this.

[image]

pico-coder, 2023-04-20:
Please read the analyzer details file.  You have likely picked a number of
channels and sample rate and number of samples that is higher than the USB
throughout.
Try less than 50k samples or sample rates under 10k to get started.   If
you still have issues get a log at debug level 3 and post it here.

pico-coder, 2024-02-05:
With no further updates, closing as likely user error trying to run the device beyond the documented capabilities.

closed 2024-02-05 by pico-coder