pico-coder/sigrok-pico #37: Request: also create a windows installer of sigrok-cli alongside pulseview

cdwijs, 2023-05-10:
I have used the windows version of pulseview mentioned here [1], and it works ok. Pulseview does have some circumstances where it forgets the captured data [2,3]

Therefore I would like to run sigrok-cli, and capture to a file, and afterwards analyze the file in pulseview. This way the captured data is saved, up to the moment sigrok-cli crashes. To this end I need a windows version of sigrok-cli, containing the driver for the pico.

Could you consider of (cross)compiling that program, the dame way it has been done with pulseview?

[1] download pulseview: https://github.com/pico-coder/sigrok-pico/blob/main/pulseview/Readme.md
[2] Request: add confirmation before starting capture when there's already captured data https://sigrok.org/bugzilla/show_bug.cgi?id=1842
[3] Request: Add auto-save while capturing https://sigrok.org/bugzilla/show_bug.cgi?id=1841

pico-coder, 2023-05-20:
I checked my windows build directories and found a sigrock-cli.exe, but unfortunately it doesn't run.  My summer vacations are starting so won't have much time to look, but I'll keep this open for when I get a chance to give a build a shot.  Might not be too bad, but trying to get the mingw etc. builds is always a little bit of fun...

cdwijs, 2023-05-20:
I've tried to compile Pulseview and sigrok-cli native in windows via MSYS2, but that fails while linking pulseview.exe:
https://sourceforge.net/p/sigrok/mailman/sigrok-devel/thread/a0f2feae5278176982fd146f14e79667%40mail.eclipso.de/#msg37844956

Next I tried to run MSYS2 in Linux under wine, but the installer freezes halfway:
https://bugs.winehq.org/show_bug.cgi?id=54938

So I'll try the recommended, cross-compile from Linux route next, as you have done.

pico-coder, 2024-02-05:
With the mainline release including this library including windows installs I'm closing this.

closed 2024-02-05 by pico-coder