pico-coder/sigrok-pico #44: pico appears & disappears, doesn't respond to sigrok/pulseview

whatotter, 2024-02-24:
(windows)

when i try to connect the pico to pulseview (pico connected, then pulseview opened) sometimes the pico doesn't appear in the serial port option of pulseview, not even device manager - this makes me skeptical if its my usb connection or the pico, but it's better to report it anyways just in case

on top of that, sometimes the pico just drops out, so every time i'd try to start reading from the pico via pulseview it'll just say "generic error", with nothing useful in the logs

also, not an issue but more of a suggestion/question, is there any support for infinite data acquisition? if i were to set the sample rate to 1mhz and set the samples to 1G it does what i ask for, but i'd think there was a simpler way...

whatotter, 2024-02-25:
got some sort of logging output:

```
sr: srpico: config_get given key 30000
sr: srpico: sample rate get of 5000
sr: hwdriver: sr_config_get(): key 30000 (samplerate) sdi 000000000826BAE0 cg NULL -> uint64 5000
sr: session: Using thread-default main context.
sr: session: Starting.
sr: hwdriver: raspberrypi-pico: Starting acquisition.
sr: srpico: Enter acq start
sr: srpico: dsbstart 3
sr: serial: Draining serial port COM8.
sr: serial: Wrote 1/1 bytes.
sr: serial: Draining serial port COM8.
sr: srpico: c 0 enabled 0 name A0 
sr: srpico: Channel enable masks D 0xFF A 0x0
sr: serial: Wrote 4/4 bytes.
sr: serial: Read 1/1 bytes.
srpico: ERROR: Serial_w_ack A00 failed (1).
sr: srpico: ack resp char $ d 36
srpico: ERROR: Channel enable fail
sr: session: Could not start raspberrypi-pico device COM8 acquisition.
sr: hwdriver: raspberrypi-pico: Stopping acquisition.
sr: srpico: At dev_acquisition_stop
sr: session: bus: Received SR_DF_END packet.
sr: serial: Read 4/32000 bytes.
sr: srpico: Dropping 4 device bytes
sr: session: Cannot remove non-existing event source 00000000068FF640.
Notifying user of session error: "Capture failed" ; "generic/unspecified error"
sr: srpico: config_get given key 30001
sr: hwdriver: sr_config_get(): key 30001 (captureratio) sdi 000000000826BAE0 cg NULL -> uint64 10
sr: srpico: config_get given key 30001
sr: hwdriver: sr_config_get(): key 30001 (captureratio) sdi 000000000826BAE0 cg NULL -> uint64 10
```

pico-coder, 2024-02-26:
So I had plans to support channel renaming, but stopped trying to implement it to try to keep my pull request more stable.
I think that if you keep the initial signal name and then add additional characters after it it might work.

pico-coder, 2024-09-07:
Please see issue 41, pretty sure this is just a duplicate

closed 2024-09-07 by pico-coder