pico-coder/sigrok-pico #45: Use of your code

palmerr23, 2024-02-26:
Hi,

I occasionally develop projects for the Silicon Chip electronics magazine, as a retirement hobby.

I thought it would be nice to develop a motherboard for sigrok-pico.

- input protection on all digital and analog inputs
- selectable 2.5 , 3.3, 5V logic levels 
- AC/DC coupling and some amplification of analog signals. one channel with 1:2:5 switched gain and the others with a simple gain pot.

Would it be OK to use your UF2 file and how-to information as part of the project? Fully acknowledged, of course!

I've attached a rendering of the initial prototype to give you an idea of the approach I'd take. Happy to share details with you, and even send you a PCB, although the magazine will have copyright over the design, so I can't open source it.

Richard
[image]

recursivenomad, 2024-08-13:
> I've never had the time to dig through all of the licensing to figure out which one I like.

@pico-coder Thanks for all your work on this project!  I'm interested in incorporating your code as a part of the firmware for a device I am developing; however, since this repository does not have a license, its source code is technically "all rights reserved" and legally prevents reuse in its current state.

I would also recommend [choosealicense.com](https://choosealicense.com) as a resource, it is very approachable.  Regarding your point `1` above, nearly any common license you choose will require that derivative works give credit to the original author, so you can consider that covered.

If you are seeking suggestions, I enjoy the simplicity and unrestricted nature of the MIT license.  However, your point `2b` above may have you lean more towards a GPL license.

A commonly cited downside to GPL is that it forces derivative works to also use the GPL, which can be argued as taking freedom away from developers.  A commonly cited downside to MIT is that it permits closed-source derivative works, which can be argued as taking freedom away from downstream users.  Certain developers may deliberately avoid reusing GPL code compared to MIT code.

*MIT and GPL-family licenses are the most commonly used licenses on GitHub (at least that was the case [back in 2015](https://github.blog/open-source/open-source-license-usage-on-github-com/)).*

palmerr23, 2024-08-13:
Thanks Peter,

I've used the MIT licence.

Pleased that you have incorporated my tweaks to your work. 

I have some spare, unpopulated boards if you'd like one. I'm not sure how we can share non-public info without a private messaging feature.

Richard

recursivenomad, 2024-08-13:
Hi Richard,

My previous message was actually intended for @pico-coder (the owner of the sigrok-pico repository) regarding licensing the firmware they've written here.  I was piggy-backing on this thread you had started about code reuse; but I'm glad you've also settled on using the MIT license for your designs 🙂

I appreciate the offer, but I do not have a need for a separate motherboard at this time.  @pico-coder did express interest earlier in this thread though, so perhaps they would still like one.

palmerr23, 2024-08-13:
Thanks Peter,

I misread the thread and will await a response from pico-coder on their continued desire for a board.

Your post also prompted me to fix up the licences on some of my own repositories.

Richard

pico-coder, 2024-09-07:
Yes @palmerr23 (Richard) I'd love to have a board.  Can you perhaps email me at the email address in the source code headers and I can forward you an address from there? 
Thanks,
Shawn

palmerr23, 2024-09-08:
Shawn,

Good to hear from you. The article has just been published in Silicon Chip magazine and I'll also send you a copy.

I looked through the headers and couldn't find your email.

Can you email me at "my github username" at gmail.com, and we'll sort things out.

Richard