pico-coder/sigrok-pico #47: capture failed  when logic channels are not selected in order

pbadel, 2024-03-28:
pulseview / sigrock-pico : waouh, great job !

i noticed one bug : when i select digital channels not in natural order (D2/D3/D4/...), the capture fails.

for example when i select only one channel : if the channel selected is D2 -> ok. if the channel selected is D3 -> not ok, capture failed

thanks again for sigrock-pico

`<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN" "http://www.w3.org/TR/REC-html40/strict.dtd">
<html><head><meta http-equiv="Content-Type" content="text/html; charset=utf-8" /><style type="text/css">
p, li { white-space: pre-wrap; }
</style></head><body>

Versions, libraries and features:
--
PulseView | 0.5.0-git-d00efc6
Qt | 5.15.8
glibmm | 2.66.5
Boost | 1_74
exprtk | 20200101
libsigrok | 0.6.0-git/4:0:0 (rt: 0.6.0-git-b503d24c-dirty/4:0:0)
- glib | 2.74.6 (rt: 2.74.6/7406:6)
- zlib | 1.2.13
- libzip | 1.7.3
- minilzo | 2.10
- libserialport | 0.1.1/1:0:1 (rt: 0.1.1/1:0:1)
- libusb-1.0 | 1.0.26.11724 API 0x01000109
- hidapi | 0.13.1
- libftdi | 1.5
- Host | aarch64-unknown-linux-gnu, little-endian
- SCPI backends | TCP, RPC, serial, USBTMC
libsigrokdecode | 0.6.0-git-0235970/4:0:0 (rt: 0.6.0-git-0235970/4:0:0)
- glib | 2.74.6 (rt: 2.74.6/7406:6)
- Python | 3.11.2 / 0x30b02f0 (API 1013, ABI 3)
- Host | aarch64-unknown-linux-gnu, little-endian
 
Firmware search paths:
/home/pb/.local/share/sigrok-firmware
/usr/local/share/sigrok-firmware
/usr/local/share/sigrok-firmware
/usr/share/raspi-ui-overrides/sigrok-firmware
/usr/share/sigrok-firmware
/usr/share/gdm/sigrok-firmware
/var/lib/menu-xdg/sigrok-firmware
/home/pb/.local/share/flatpak/exports/share/sigrok-firmware
/var/lib/flatpak/exports/share/sigrok-firmware
/usr/local/share/sigrok-firmware
/usr/share/sigrok-firmware
 
Protocol decoder search paths:
/usr/local/share/libsigrokdecode/decoders
/usr/local/share/libsigrokdecode/decoders
/usr/local/share/libsigrokdecode/decoders
(Note: Set environment variable SIGROKDECODE_DIR to add a custom directory)
 
Supported hardware drivers:
asix-omega-rtm-cli | ASIX OMEGA RTM CLI
asix-sigma | ASIX SIGMA/SIGMA2
beaglelogic | BeagleLogic
chronovu-la | ChronoVu LA8/LA16
demo | Demo driver and pattern generator
dreamsourcelab-dslogic | DreamSourceLab DSLogic
ftdi-la | FTDI LA
fx2lafw | fx2lafw (generic driver for FX2 based LAs)
greatfet | Great Scott Gadgets GreatFET One
gwinstek-gds-800 | GW Instek GDS-800 series
hameg-hmo | Hameg HMO
hantek-4032l | Hantek 4032L
hantek-6xxx | Hantek 6xxx
hantek-dso | Hantek DSO
hung-chang-dso-2100 | Hung-Chang DSO-2100
ikalogic-scanalogic2 | IKALOGIC Scanalogic-2
ikalogic-scanaplus | IKALOGIC ScanaPLUS
ipdbg-la | IPDBG LA
kingst-la2016 | Kingst LA2016
lecroy-logicstudio | LeCroy LogicStudio
lecroy-xstream | LeCroy X-Stream
microchip-pickit2 | Microchip PICkit2
ols | Openbench Logic Sniffer & SUMP compatibles
p-ols | Pipistrello OLS
raspberrypi-pico | RaspberryPI PICO
rigol-ds | Rigol DS
saleae-logic-pro | Saleae Logic Pro
saleae-logic16 | Saleae Logic16
siglent-sds | Siglent SDS1000/SDS2000
sysclk-lwla | Sysclk LWLA series
sysclk-sla5032 | Sysclk SLA5032
yokogawa-dlm | Yokogawa DL/DLM
zeroplus-logic-cube | ZEROPLUS Logic Cube LAP-C series
 
Supported input formats:
binary | Raw binary logic data
chronovu-la8 | ChronoVu LA8/LA16 native file format data
csv | Comma-separated values
logicport | Intronix LA1034 LogicPort project
null | Null input (discards all input)
protocoldata | Generate logic traces from protocol's data values
raw_analog | Raw analog data without header
saleae | Saleae Logic software export files
stf | Sigma Test File (Asix Sigma/Omega)
trace32_ad | Lauterbach Trace32 logic analyzer data
vcd | Value Change Dump data
wav | Microsoft WAV file format data
 
Supported output formats:
analog | ASCII analog data values and units
ascii | ASCII art logic data
binary | Raw binary logic data
bits | 0/1 digits logic data
chronovu-la8 | ChronoVu LA8 native file format data
csv | Comma-separated values
hex | Hexadecimal digits logic data
null | Null output (discards all data)
ols | OpenBench Logic Sniffer data
srzip | srzip session file format data
vcd | Value Change Dump data
wav | Microsoft WAV file format data
wavedrom | WaveDrom.com file format
 
Supported protocol decoders:
ac97 | Audio Codec '97
ad5626 | Analog Devices AD5626
ad79x0 | Analog Devices AD79x0
ade77xx | Analog Devices ADE77xx
adf435x | Analog Devices ADF4350/1
adns5020 | Avago ADNS-5020
adxl345 | Analog Devices ADXL345
am230x | Aosong AM230x/DHTxx/RHTxx
amulet_ascii | Amulet LCD ASCII
arm_etmv3 | ARM Embedded Trace Macroblock v3
arm_itm | ARM Instrumentation Trace Macroblock
arm_tpiu | ARM Trace Port Interface Unit
atsha204a | Microchip ATSHA204A
aud | Advanced User Debugger
avr_isp | AVR In-System Programming
avr_pdi | Atmel Program and Debug Interface
caliper | Digital calipers
can | Controller Area Network
cc1101 | Texas Instruments CC1101
cec | HDMI-CEC
cfp | 100 Gigabit C form-factor pluggable
cjtag | Compact Joint Test Action Group (IEEE 1149.7)
counter | Edge counter
dali | Digital Addressable Lighting Interface
dcf77 | DCF77 time protocol
dmx512 | Digital MultipleX 512
ds1307 | Dallas DS1307
ds2408 | Maxim DS2408
ds243x | Maxim DS2432/3
ds28ea00 | Maxim DS28EA00 1-Wire digital thermometer
dsi | Digital Serial Interface
edid | Extended Display Identification Data
eeprom24xx | 24xx I²C EEPROM
eeprom93xx | 93xx Microwire EEPROM
em4100 | RFID EM4100
em4305 | RFID EM4205/EM4305
enc28j60 | Microchip ENC28J60
flexray | FlexRay
graycode | Gray code and rotary encoder
guess_bitrate | Guess bitrate/baudrate
hdcp | HDCP over HDMI
i2c | Inter-Integrated Circuit
i2cdemux | I²C demultiplexer
i2cfilter | I²C filter
i2s | Integrated Interchip Sound
ieee488 | IEEE-488 GPIB/HPIB/IEC
ir_irmp | IR IRMP
ir_nec | IR NEC
ir_rc5 | IR RC-5
ir_rc6 | IR RC-6
ir_sirc | Sony IR (SIRC)
jitter | Timing jitter calculation
jtag | Joint Test Action Group (IEEE 1149.1)
jtag_ejtag | Joint Test Action Group / EJTAG (MIPS)
jtag_stm32 | Joint Test Action Group / ST STM32
lfast | NXP LFAST interface
lin | Local Interconnect Network
lm75 | National LM75
lpc | Low Pin Count
ltc242x | Linear Technology LTC242x
ltc26x7 | Linear Technology LTC26x7
maple_bus | SEGA Maple bus
max7219 | Maxim MAX7219/MAX7221
mcs48 | Intel MCS-48
mdio | Management Data Input/Output
microwire | Microwire
midi | Musical Instrument Digital Interface
miller | Miller encoding
mlx90614 | Melexis MLX90614
modbus | Modbus RTU over RS232/RS485
morse | Morse code
mrf24j40 | Microchip MRF24J40
mxc6225xu | MEMSIC MXC6225XU
nes_gamepad | Nintendo Entertainment System gamepad
nrf24l01 | Nordic Semiconductor nRF24L01(+)
nrf905 | Nordic Semiconductor nRF905
numbers_and_state | Interpret bit patters as numbers or state enums
nunchuk | Nintendo Wii Nunchuk
onewire_link | 1-Wire serial communication bus (link layer)
onewire_network | 1-Wire serial communication bus (network layer)
ook | On-off keying
ook_oregon | Oregon Scientific
ook_vis | On-off keying visualisation
pan1321 | Panasonic PAN1321
parallel | Parallel sync bus
pca9571 | NXP PCA9571
pjdl | Padded Jittering Data Link
pjon | PJON
ps2 | PS/2
pwm | Pulse-width modulation
qi | Qi charger protocol
rc_encode | Remote control encoder
rfm12 | HopeRF RFM12
rgb_led_spi | RGB LED string decoder (SPI)
rgb_led_ws281x | RGB LED string decoder (WS281x)
rtc8564 | Epson RTC-8564 JE/NB
sae_j1850_vpw | SAE J1850 VPW.
sbus_futaba | Futaba SBUS (Serial bus)
sda2506 | Siemens SDA 2506-5
sdcard_sd | Secure Digital card (SD mode)
sdcard_spi | Secure Digital card (SPI mode)
sdq | Texas Instruments SDQ
seven_segment | 7-segment display
signature | Signature analysis
sipi | NXP SIPI interface
sle44xx | SLE44xx memory card
spdif | Sony/Philips Digital Interface Format
spi | Serial Peripheral Interface
spiflash | SPI flash/EEPROM chips
ssi32 | Synchronous Serial Interface (32bit)
st25r39xx_spi | STMicroelectronics ST25R39xx
st7735 | Sitronix ST7735
stepper_motor | Stepper motor position / speed
swd | Serial Wire Debug
swim | STM8 SWIM bus
t55xx | RFID T55xx
tca6408a | Texas Instruments TCA6408A
tdm_audio | Time division multiplex audio
timing | Timing calculation with frequency and averaging
tlc5620 | Texas Instruments TLC5620
uart | Universal Asynchronous Receiver/Transmitter
usb_packet | Universal Serial Bus (LS/FS) packet
usb_power_delivery | USB Power Delivery
usb_request | Universal Serial Bus (LS/FS) transaction/request
usb_signalling | Universal Serial Bus (LS/FS) signalling
wiegand | Wiegand interface
x2444m | Xicor X2444M/P
xfp | 10 Gigabit Small Form Factor Pluggable Module (XFP)
xy2-100 | XY2-100(E) and XY-200(E) galvanometer protocol
z80 | Zilog Z80 CPU
 
Available Translations:
German | (Sören Apel, Uwe Hermann)
Spanish | (Carlos Diaz, Ulices Avila Hernandez)
Japanese | (Yukari Shoji)
Chinese | (ZtyPro)

</body></html>
`

pico-coder, 2024-03-28:
Per https://github.com/pico-coder/sigrok-pico/blob/main/AnalyzerDetails.md
"Digital channels are PICO board pins D2-D22 and are named accordingly in sigrok. Channels must be enabled (via Pulseview or sigrokcli) starting at D2 and continuously set towards D22"

This limitation exists because it greatly simplifies the compression of data across USB.

closed 2024-03-28 by pico-coder

pbadel, 2024-03-28:
my bad ! read the doc too superficially