pico-coder/sigrok-pico #48: 5x Faster ADC

palmerr23, 2024-04-02:
I found this recent post in the R Pi forums.

https://forums.raspberrypi.com/viewtopic.php?t=365702

I managed to get the fastest version running with a single ADC channel working with very little fuss at a 240MHz clock rate. ENOB still appears to be > 7-bits. I haven't tried multi-channel ADC, but can't see that the Analog Mux is likely to pose any issues.

5x faster ADC would make a very nice improvement!

palmerr23, 2024-04-03:
One simple change made it (partially) work:

Setting the ADC clock source in the CLK_ADC_CTRL register to the system clock rather than the ADC's usual 48MHz clock, immediately after adc_init() (around line 715 in pico_sdk_sigrok.c)

```
#define ADC_OCLK_SRC ((uint32_t*)0x40008060)
adc_init();
*ADC_OCLK_SRC = 0x820; //RP
```

The ADC sample rate increased by a factor of 2.5 without changing any of the clk_div code.

So far I have only tested 100 samples at 250kHz (and 500kHz) with a 60kHz sine wave as input. We get 2.5x as many samples per cycle, however the displayed sine wave's period is off by the same factor. More work to do on this!

Both images below are at 250kHz and ~400uS long.

Fast clock:
[image]

Original clock:
[image]

palmerr23, 2024-04-03:
Changing the two uses of a 48MHz clock to 240MHz in the ADC clock divider code, seems to have resolved the issue.

```
#define ADC_CLK_SPEED 240000000ULL // 48000000ULL
~938                uint32_t adcdivint=ADC_CLK_SPEED/(dev.sample_rate*dev.a_chan_cnt);
~960                adc_frac_int=(uint8_t)(((ADC_CLK_SPEED%dev.sample_rate)*256ULL)/dev.sample_rate);
```

2.4Mhz sample rate, 100 samples, 42uS total capture, input 60kHz sine wave:
[image]

TheCustomFHD, 2024-07-14:
This, plus running the flash clock divider at 4 instead of the default 2, should allow clockspeeds of 450MHz or more. My particular Pico runs at 450MHz stable at 1.3V

pico-coder, 2025-10-15:
This isn't forgotten about, but I spent the bulk of my development time doing a large DMA rewrite and adding RP2350 support.  I do need to also get some libsigrok/pulseview updates in for that, but after that this is probably highest priority.