pico-coder/sigrok-pico #49: ngscopeclient port

sslupsky, 2024-04-13:
Have you considered porting to ngscopeclient?
https://www.ngscopeclient.org

pico-coder, 2024-04-13:
The limitation of many of the supported protocols is that they implement a fixed sample size from the host which the device then responds with.  For a real piece of hardware with lots of on board storage that's no big deal, but the PICO only has ~200kB of useable space.  And it's across a slow 12mb/s usb port.  Thus I implemented a dynamic compression algorithm for digital samples to enable things like 400ksps i2c that could stream up to the memory limits of the host (GBs) rather than the the device.
You could go to more standard SCPI protocol, but you'd lose that capability.
Also note that ngscopeclient doesn't support windows SCPI so it would be limited to use.

If I really was going to go to another protocol, I think I'd develop some javascript code that used either WebSerial API or used a wireless pico over ethernet and then displayed on a javascript canvas.  I think now that's it 2024 just about any modern machine should render that just fine, and the javascript should be able to parse out the serial stream too. Then, any time someone wants a new feature they just need to write some javascript.  That eliminates lots of fussing across multi-OS build issues with all of the components that libsigrok uses.

sslupsky, 2024-04-13:
Yeah, that makes sense.  I pinged the guys over at the ngscopeclient side to ask if they intend to support streaming.  There is an open issue about this but no indication other than it is an open issue.

There is some work going on to support an open source scope with a Thunderbolt interface.  Not sure if any of that work considered streaming data or if it implements the interface in a more traditional manner.

https://github.com/EEVengers/ThunderScope

sslupsky, 2024-04-13:
Looks like they use SCPI transport for the Thunderscope.

pico-coder, 2024-04-14:
Yeah, the RP2040 is an amazing little device, but in many ways, it's just a little device.  I've kind of had the philosophy that it's a logic analyzer and scope that you use when you don't have anything else around to do more serious debug...

sslupsky, 2024-08-22:
The new RP2350 now supports read and write over the QSPI bus and has two chip selects so you can attach a PSRAM to the QSPI bus.  Might be something to think about for a larger on board buffer.

pico-coder, 2024-09-07:
Closing this issue as ngscope client support won't be supported.
As for the RP2350, I really wish they had added USB 2.0 support (480MBit/sec) as that would have been a huge bandwidth improvement.
The QSPI bus support is interesting, but by the time you buy the flash ram and make a board to put on it, you might as well just by a Salae clone for cheaper and get USB 2.0 support.

closed 2024-09-07 by pico-coder

sslupsky, 2024-09-08:
The RP2350 has TMDS encoders.  Could you implement a DPI or HDMI interface and utilize the HDMI bandwidth?  You would need some sort of HDMI to USB converter ... an HDMI video capture dongle could provide the HDMI to USB C port connection to the computer.

sslupsky, 2024-09-08:
Adafruit sells an RP2040 board with an HDMI connector.

https://www.adafruit.com/product/5710

sslupsky, 2024-09-08:
Could use this DVI breakout board with an RP2350

https://www.adafruit.com/product/4984

sslupsky, 2024-09-08:
Here is a repo for running DVI on a Pico

https://github.com/Wren6991/PicoDVI

sslupsky, 2024-09-08:
Interesting discussion here about using DP Alt mode across a USB C connector for the high bandwidth back channel:
https://news.ycombinator.com/item?id=41260679