pico-coder/sigrok-pico #50: Driver Problems

PMTGaming, 2024-06-27:
Hello,

I have transferred the pico_sdk_sigrok.uf2 file onto the Raspberry Pico and installed the PulseView version from the Dropbox link provided in the README. I used Zadig to install the USB serial CDC driver, but I am encountering the following error: "The device cannot start. (Code 10)."

Additionally, it states: "A device which does not exist was specified."

[image]

[image]

pico-coder, 2024-07-17:
I updated Readme.md to mention that the dropbox link and my windows builds instructions should be avoided, and the mainline release at sigrok.org/wiki/Downloads should be used.
I also added some possible things to try in https://github.com/pico-coder/sigrok-pico/blob/main/GettingStarted.md when the device doesn't work on windows.
However, most of the serial stuff is out of my control, as it is part of windows, or the libserial library etc.  Further, I have no way of debugging serial device failures under windows.  Usually plugging/unplugging/rebooting eventually works.

pico-coder, 2024-09-07:
Assuming these are the usual issues with getting device connected the first time.  Closing.

closed 2024-09-07 by pico-coder