pico-coder/sigrok-pico #53: Feature Request: Disable debug UART and make pins available

ereisch, 2024-11-13:
If this code is relatively stable, it would be nice (if it's even possible; I'm not a RPi engineer) if the 4 GPIO ports allocated for the debug UART were able to be allocated as additional digital inputs.  Putting the device at or over the 24-channel barrier would greatly increase its utility, as you could then fully analyze embedded controllers with a 16-bit address and 8-bit data bus.  A 25th input would be even better, since that could be tied to either a clock signal, a R/!W signal, or something similar.

I also wonder if you'll save CPU cycles and be able to successfully have a sample-on-rising-edge feature if the ADCs and debug printf()s were omitted (since printf() is a very expensive operation).  Not having time to look into the code at the moment, are interrupts being set on trigger pins?  This could save the overhead of having to software poll by just having the hardware interrupt logic watch a pin for either rising or falling edge, and then the ISR samples all the GPIO channels.

Just a thought.

Schwuuuuup, 2025-08-11:
For my current project I's ideally need 32 Pins and if not this , than the more digital pins, the better. The PIO width is 32 as I understand, so this seems to be a limit anyway.
The Option for the ADCs is great, when you need them, but they seem to consume so much bandwidth, that faster captures are not viable.
So I for one would welcome a digital-only version. So any enlightening lines on how one coul achieve such a "special build mode" would be great.

An one more thought: I acually have a Waveshare rp2440-zero here, which has an addressable RGB-LED on Pin 16  (so it is not usable as an input, but it has D25 and D29 (ADC4) instead.
Unfortunately I'm not good at C and have no experience with rp2040 based boards, so my gasp on how your code works is very limited at the moment

pico-coder, 2026-09-14:
https://github.com/pico-coder/sigrok-pico/issues/55 has comments in the Sept 2026 timeframe that the UART may be the cause of some user issues with USB reliability.  In theory the two should be able to coexist, but for some users it doesn't work:
Possible theorys:
- High rates of noise create constant interrupts for the uart.
- Some kind of incompatibility with mixing uart and cdc that conflicts, especially because this repo doesn't directly use stdio but instead uses lower level functions that the pico SDK doesn't support or envision.
This will require debug information to be sent mainband which will require libsigrok driver changes.

dougie996, 2026-09-14:
Thanks for taking your time!  

I tried on my AliExpress Gusman V6 Analyzer Board:

pico2_baseline
pico2_rel3_baseline.uf2
pico_sdk_sigrok_rp2350b.uf2
pico_sdk_sigrok_pico2.uf2
pico_baseline.uf2

None of those worked for me, so I hooked myself up to the debug UART and saw it was hanging in a receive loop.

Just disabling the Rx and patching it to 2.3.1 did the job for me. Running fine now with Pulseview