pico-coder/sigrok-pico #54: Sigrok detects the device but PulseView does not

FedericoCeratto, 2024-11-23:
Hello and thanks for developing pico-coder.
I'm not quite sure if this is a driver issue or not, please direct me to a different bugtracker if needed.

Sigrok seems to be able to detect the device:
```
./sigrok-cli-NIGHTLY-x86_64-debug.AppImage -d ols:conn=/dev/ttyACM2 --show
Driver functions:
    Logic analyzer
Scan options:
    conn
    serialcomm
    probe_names
ols:conn=/dev/ttyACM2 - uLA: Micro Logic Analyzer with 16 channels: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15
Supported configuration options:
    conn: /dev/ttyACM2 (current)
    Maximum number of samples: 100000
    samplerate (10 Hz - 100 MHz in steps of 1 Hz)
    Supported triggers: 0 1
    captureratio: 0 (current)
    external_clock: on, off (current)
    clock_edge: rising (current), falling
    pattern: None (current), External, Internal
    swap: on, off
    rle: on, off (current)
```

PulseView 0.4.2 detects the tty port in use correctly but when I try to scan for devices it fails with:
```
sr: srpico: Drain reads
sr: srpico: Drain reads done
sr: srpico: ERROR: Serial_w_resp failed (0).
sr: srpico: 1st identify failed
sr: srpico: Send second *
sr: srpico: ERROR: Serial_w_resp failed (0).
sr: srpico: Second attempt failed
```

(I tried resetting the device and restarting PulseView)

Thanks

pico-coder, 2024-11-28:
Your sigrok-cli is finding the wrong device.   First do an lsusb to make sure the PICO is being found, and then a sigrok-cli list-serial

Here is what it looks like for the pico:
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2e8a:000a Raspberry Pi Pico
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

 $ sigrok-cli --list-serial
Available serial/HID/BT/BLE ports:
  /dev/ttyACM0	Pico - E660C06213643B29

 $ sigrok-cli -l 2 -d raspberrypi-pico:conn=/dev/ttyACM0:serialcomm=115200/flow=0 --scan
sr: srpico: Drain reads
sr: srpico: Drain reads done
The following devices were found:
raspberrypi-pico - Raspberry Pi PICO 00 [S/N: N/A] with 24 channels: A0 D2 A1 D3 A2 D4 D5 D6 D7 D8 D9 D10 D11 D12 D13 D14 D15 D16 D17 D18 D19 D20 D21 D22

pico-coder, 2025-03-19:
With no further updates I'm closing this assuming it's the usual "getting the first connection is more difficult than it seems".

closed 2025-03-19 by pico-coder