pico-coder/sigrok-pico #56: Pinout assignment of RasPi Pico

mumbaiverve, 2024-12-28:
Hi,
Is there a doc which maps the GPIO of the RaspPi Pico to the PulseView GUI ? For eg. GP0 corresponds to which channel on PulseView ?

pico-coder, 2024-12-29:
The "Dx" pins map to the "GPx" pins of the Pico board , and "Ay" pins map to the ADCy pins of the board.

pico-coder, 2024-12-29:
See https://github.com/pico-coder/sigrok-pico/blob/main/AnalyzerDetails.md

mumbaiverve, 2024-12-30:
Hi,
Thank you for replying. This is solved.
Just found  a couple of minor annoyances which threw me for some time 
1) Pulse View does not connect to the device if we select a baud rate. If we leave the dropdown blank, then it connects all the time perfectly.
2) If we select channels randomly it throws a error ("capture failed generic unspecified error") and does not work. If we select channels in sequence , or all channels then it works fine.
Thanks again for your work.