pico-coder/sigrok-pico #58: speed of CDC

rapzak, 2025-01-12:
Hi,

What speed do you have od CDC, as i do not see you have increased buffers for usb?
I managed to get 600kbyte/sec instead 195kbyte/sec after increased buffers...
Take a look:
https://forums.raspberrypi.com/viewtopic.php?p=2285416#p2285416

Great project.. i am doing something simulary with 4 external ADC... higher resolution but slower samplerate...

Cheers
Kasper

rapzak, 2025-01-12:
Can be closed - just info

rapzak, 2025-01-12:
not sure if visible if closed - you can just close it when read...

Kalki2898add, 2025-03-01:
i uploaded the code in Rp2040 

[image]

but pulse view not detecting any serial port 
how to fix the problem?

pico-coder, 2025-03-20:
@rapzak , I tested the release uf2 on my windows machine, enabling 3 ADC channels and 1 digital channel.  That's 4Bytes of data per sample.  I then set a number of samples (500ksamples * 4 bytes per sample) much greater than the memory of the device (256).  Streaming worked at 160Ksps sample rate, but failed at 200Ksps.  So I'm hitting 640KB/sec, but not reaching 800KB/sec
Note that generally speaking more buffer space doesn't necessarily mean more bandwidth, as I think the SDK already has plenty of buffer space to utilize the link.  But what is key is minimizing the calls to the CDC code and limiting other overheads that keep the buffering from being full.  That's why I wrote the  my_stdio_usb_out_chars function to allow more direct control of buffering.
Though for full disclosure, I did do quite a bit of tweaking of USB related coded and did that on a very old version of the SDK, so things may have changed since.  Though at a 12Mbps speed of USB, that's a max value 1.5KB/sec, and if you consider that USB 2.0 doesn't have dedicated Rx and Tx lines, I may be fairly close to maximum.

I'm going to start work on some new feature adds which will require me to update to the new SDK. I'll leave this open as a reminder to make sure I continue to see that bandwidth rates.   I'll also see if the raspberry pi forum link is appropriate, but I'm guessing that the change is only needed for cases where the user is sending large chunks of data all at once to the CDC.   With the better buffer management already in this codebase having more buffer space probably doesn't help.