pico-coder/sigrok-pico #61: pulse view not detecting Raspberry pi pico driver

Kalki2898add, 2025-03-01:
i uploaded the uf2 then its shows the driver 

[image]

but in pulse view software there is no driver detecting. 

[image]

i tried every thing but not working on windows10
but when i tried on linux debian its working,
please tell me solutions

BerranRemzi, 2025-03-01:
On my first attempt with sigrok-pico, I encountered a similar issue, so you can try changing the USB drivers using Zadig.

Kalki2898add, 2025-03-02:
> On my first attempt with sigrok-pico, I encountered a similar issue, so you can try changing the USB drivers using Zadig.

can you send me the picture of your port on device manager?

Nomis91, 2025-03-12:
I have the same issue.
When rebooting the pc (win10 22H2) the pico does work once.
If I disconnect PulseView and disconnect the pico und reconnect it again. the pico will not be found by PulseView. But its present in the hardware manager. with a com port and a windows driver:

[image]

I tried to open a connection with HTerm but there comes no response from the pico.

I tried to change the driver with Zadig to WinUSB(v6.1....) but Zadig does not show the Serial Device. Only the reset device is shown.

[image]

pico-coder, 2025-03-12:
Aside from what is in https://github.com/pico-coder/sigrok-pico/blob/main/GettingStarted.md I don't have much to add.
The serial connection is sometimes unexplainably difficult for some and then it just starts working..
Do make sure that you only have one application open at a time when accessing a serial device, as Windows will often "lock" one app to a COM port.
You might close all serial apps including pulseview and see if Terraterm/putty can connect with a device.  Send a 'i' and see if a string comes back.

Nomis91, 2025-03-12:
After experimenting the last 5 hours I got it working using following settings:
1.: zadig -> show all devcies -> select device according to id in device explorer (names does not match) -> install CDC driver -> close zadig 
2.: use putty to establish connection first -> send i -> got response -> close putty
3.: open PulseView -> connect to pico -> have fun

after disconnecting pico steps 2 + 3 have done again.

hTerm does not work for me.

pico-coder, 2025-03-20:
thanks for the zadig details, it's been a while since I've ran it.  I started Zadig myself and the trick (as you mention) is
Options-> List All Devices
From the drop down look for something like "Board CDC (Interface 0)"  (or similarly, run Zadig without the device plugged in, then plug it in, restart zadig and see what device is new.  Then select that device.
Under the driver, select "USB Serial CDC" and then install that driver.
I need to updates the getting started page with this info, and then will close this out.