pico-coder/sigrok-pico #62: Support more universal pinouts / 30 GPIOs

c1570, 2025-03-19:
How about getting rid of the two UART pins (no need to waste those as they are only necessary for debugging)?

Also, there are RP2040/2350 boards that expose all GPIOs (e.g., Pimoroni PGA2040 or the violet "Ultimate" boards available on Aliexpress). With those, 30 digital channels would be possible (if not using the shared analog inputs).
However, the `lval<<=11, lval>>=11` dance in the firmware disallows that.
Wouldn't it be better to put that masking in libsigrock (why is it even necessary - because of RLE?)?

pico-coder, 2025-10-15:
I just pushed a major code release, and in the release directory are new uf2s with "dig26" or "dig32" in their name. 
Note that pulseview is currently hardcoded to start at "D2" even though D0 and D1 in these modes are digital inputs (eventually I'll get a pulseview commit going, but one thing at a time...).  So you'll see D2 to D27 rather than D0 to D25.
Please try those out and let me know how it goes....

c1570, 2025-10-15:
Nice!

Originally I planned to add support for 48 GPIOs using the RP2350B and already modified libsigrok for that, too, see https://github.com/c1570/libsigrok/commits/rppico_64channels/ . However I got distracted by other projects and didn't update the RP2350B side so far - which is a bit more involved since for 48 GPIOs two running PIOs are needed as each PIO is limited to 32 GPIOs...

I also started writing some rp2350 emulator based testing code using https://github.com/c1570/rp2040js (testing code works but isn't finished though)

Note that https://github.com/c1570/libsigrok/commit/9d2af3bd97e63ad2f9702b1f9281a60c58e3f97e fixes a libsigrok/pico bug (I think) - without that, the device wouldn't recover from buffer overflow failure state.

Will try the UF2, it'll take some time though...