pico-coder/sigrok-pico #66: Appimage aborts on Fedora41

bitwisetech, 2025-06-22:
Hello, 

  I've downloaded the Appimage and made it executable, on one Fedora41 machine pulseview starts OK,  on the machine 
that I need to run, it fails: 

/opt<=>./pulseview-NIGHTLY-x86_64-debug.AppImage 
ASSERT: "!isEmpty()" in file /opt/qt512/include/QtCore/qlist.h, line 347
Aborted (core dumped)

On the 'good' machine, running 'locate qlist.h' results in: 
/usr/include/qt5/QtCore/qlist.h

On the 'bad' machine I see: 

/usr/include/Qt/qlist.h
/usr/include/QtCore/qlist.h

But I don't know if there is a package I must install.  

Thanks for any help !

pico-coder, 2025-09-08:
Sorry I got behind over the summer.  I do all my linux work on pulseview on a raspberry pi with raspberry PI OS and I do native builds on the device.  So none of those directories exist for me.  Yes, you likely need some kind of Qt package, but I would have thought that an AppImage would include that...  I once tried doing appimages and Qt was a persistent pain that I never resolved.

pico-coder, 2025-10-15:
I don't have anything else to add, and really this is a pulseview build problem, not an issue with the sigrok-pico driver or device code, so closing.  Good luck.  As somehow who's done pulseview builds you have my sympathies.